Ultimaker/Cura #11382: Support structure is generated in the part

brandlc, 2022-01-30:
### Application Version

4.13.0

### Platform

Windows 10

### Printer

Formbot T-rex 3

### Reproduction steps

For components that need to generate hollow and support structures inside, support structure is also generated inside the component. The error only occurs with tree supports.

### Actual results

[image]
[image]
[image]

### Expected results

Support structures should not be printed in the component, but on top of it if necessary.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]
[image]

GregValiant, 2022-01-30:
There is an emergency Cura release to fix this.  @fvrmr's post on the [ULTIMAKER FORUM](https://community.ultimaker.com/topic/39459-ultimaker-cura-4131/#comment-302241) has the download link to 4.13.1.

fvrmr, 2022-01-31:
Hi @brandlc thank you for your report.
This is indeed fixed in 4.13.1 and this issue is a duplicate of https://github.com/Ultimaker/Cura/issues/11258
You can download 4.13.1 here: https://ultimaker.com/software/ultimaker-cura
I will close this issue.

closed 2022-01-31 by fvrmr