Ultimaker/Cura #11384: slicing failed

justinreed270, 2022-01-30:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

I imported a STL file. 

I clicked slice.

The slice error message appeared. 

### Actual results

message failure

### Expected results

The file should be slices and I should be able to send it to my Ender 3 via OctoPi server.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/a

fvrmr, 2022-01-31:
Hi @justinreed270 thank you for your report.
Could you share your project file (file - save project) as a .zip and your log file: `C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`
Without these files it is difficult to know what could cause this.

justinreed270, 2022-01-31:
I wish I would have known that you needed this information...

I ended up uninstalling and reinstalling Cura... This fixed the issue. Next
time, I will save the logs and project in a .zip file and send it to your
company.

Thanks,

Justin Reed

731.592.0565

fvrmr, 2022-02-01:
Hi @justinreed270 thank you for your reply.
Good to know that the issue is fixed for you.
I will close this issue, but reach out if it happens again!

closed 2022-02-01 by fvrmr