Ultimaker/Cura #11385: Cura Crashes 100% of the time when changing views between multiple Octoprint instances within monitor view.

mcclinsr, 2022-01-30:
### Application Version

4.13.0

### Platform

MacOS 12.2, Windows 10, Windows 11

### Printer

Creality Ender 3 (multiple)

### Reproduction steps

Toggle to "Monitor" mode in Cura, view live print in progress (Octoprint with camera) switch view from printer 1 to printer 2.

### Actual results

Cura software force closes and crashes. 

### Expected results

seamless transition between printer 1 and printer 2 to preview the cameras.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This is true of multiple versions of Cura (I have been using Cura since 4.9 but could have been an issue prior). I use both MacOS and a windows machine that began on Windows 10 and upgraded to Windows 11 (none of which fixed the issue).

fvrmr, 2022-02-01:
Hi @mcclinsr thank you for your report.
Could you share your log file, for MacOS you can find your log file here: `$HOME/Library/Application Support/cura/<Cura version>/cura.log` and Windows here: `C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

mcclinsr, 2022-02-01:
Here is the log. I made an effort to replicate the issue multiple times
before sending over the logs. It appears to happen most often when the
printers are active (I have 2) and I specifically switch between the one
named "L.Ender" to the one named "B.Ender" while in the Monitor tab. in
this scenario it happens 100% of the time. If I toggle the other direction
(from B.Ender to L.Ender) it happens some of the time.

fvrmr, 2022-02-01:
I can't see your log file since you replied via e-mail. Please attach in a comment here on Github.

fieldOfView, 2022-02-01:
This is likely the same issue as https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/190, which seems to be an issue in QtQuick controls 1. The print monitor is currently being rewritten with QtQuick Controls 2 for the next version of Cura, which does not have this problem.

mcclinsr, 2022-02-01:
[cura.log](https://github.com/Ultimaker/Cura/files/7981581/cura.log)

Here is the log file attached to Github

mcclinsr, 2022-02-02:
> This is likely the same issue as fieldOfView/Cura-OctoPrintPlugin#190

I uploaded my logs to the related github. Can you use these logs to confirm
if that is the issue and if so, what the timeline is for the resolution
code? This error has been apparent for some time so I am eager to find a
fix.

Thank you.

fvrmr, 2022-02-11:
Since this is related to the Octoprint plugin like @fieldOfView mentioned, I will close this issue. Please continue the thread in the other issue.

closed 2022-02-11 by fvrmr