Ultimaker/Cura #11388: Slicing adds bottom layer where a brim should be

GunnardVS, 2022-01-31:
### Application Version

4.13.0

### Platform

Windows 10

### Printer

Creality Ender 5 Plus

### Reproduction steps

1. I used my standard profile (closing holes is off, Union overlapping Volumes can be on or off, doesn't matter)
2. loaded an STL of the top of a Dice tower, which has a dome like structure and a wall half the way, opposite the opening to throw in the die. (see pictures)
3. slice the model
4. wonder about the result, because 4.9 and 4.10 and 4.11 sliced correctly - Prusaslicer also slices correctly (see third picture)

[image]

### Actual results

CURA adds a bottom layer for half of the dome and closes all three peg holes during this (see second picture)
[image]

### Expected results

Correctly sliced model with brim on the inside and outside, like the screenshot from PrusaSlicer, I had to use instead.

[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Please have a look at the attached project file

[CE5P_DruidRoof_DT_roof_only_v1.0.3mf.zip](https://github.com/Ultimaker/Cura/files/7970112/CE5P_DruidRoof_DT_roof_only_v1.0.3mf.zip)

GregValiant, 2022-01-31:
"Make Overhangs Printable" is making the overhangs printable.  Turn it off.
I would suggest turning off "Coasting" as well.  With that .8 nozzle the pressure will really fall off at the end of an extrusion.

GunnardVS, 2022-01-31:
I'm just confused, because I didn't change the overhang setting between printing this STL the first two times and now. 
I just dropped the speed a bit.
And adding several bottom layers to an actually empty space doesn't make any overhang more printable - the same goes for just closing the peg holes. 
So there is a very good reason, this function is "experimental"
The result looks quite good with the .8 Nozzle - except for the very top, where it becomes a collection of blobs on the surface - the original reason to turn on coasting (the difference between print 1 and 2).
Hiding the seam might be a culprit here... it makes sense on the big bits, but close to the top, it rather looks like a blobby mess.

The Slicing in CURA makes it look much worse than it really is in reality (except for the top).

closed 2022-01-31 by GunnardVS