Ultimaker/Cura #11389: lines on the printout, there is no such thing on the stable markup

maciej123m, 2022-01-31:
### Application Version

Cura Arachne beta 2

### Platform

Windows 10

### Printer

Ender 3 Custom

### Reproduction steps

I wanted to print a voron calibration cube
[image]

### Actual results

Lines appear on the printout as if the filament was not given, it only happens in specific places where the head is moved
[image]

[image]

### Expected results

the cube should be as smooth as that printed on the stable version
[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CE3_Voron_Design_Cube_v7.zip](https://github.com/Ultimaker/Cura/files/7970272/CE3_Voron_Design_Cube_v7.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/7970273/cura.log)

fvrmr, 2022-02-01:
Hi @maciej123m thank you for your report.
Did you use the same project file in Arachne and 4.13?

maciej123m, 2022-02-01:
Yes, I used the same design on both versions of cura

fvrmr, 2022-02-04:
Thank you for your response. I think this is related to the seam. In Arachne the position of the seam is not applied correctly (see https://github.com/Ultimaker/Cura/issues/10181) 

4.13.1:
[image]

And if I look in Arachne you see on layers: 21, 30, 60, 90, 91 the seam in another place:
[image]

And these layers look like the lines you are seeing in your model. When using the playback per layer the direction of these layers are different.

Ghostkeeper, 2022-02-15:
Indeed, good analysis.

It doesn't place the seam in the same corner on those layers because the wall actually turned into a polyline instead of a loop. On the inside it's encountering a 3-way joint and the outside wall happens to be chosen as the 3rd wheel, as it were. Since polylines have two endpoints, it (correctly) chooses to start from one of those instead of to start from the seam.

I think this will be hard to fix, but we do have the intention to improve the stitching before the release.