Ultimaker/Cura #11390: upgrade old developed function

tmirante, 2022-01-31:
### Is your feature request related to a problem?

i was not able ( my inabilty) to upgrade the extension i developed on a cura  4.3.0 old version at last version. 

### Describe the solution you'd like

i belive this will solve hotend tuning on printer i3 anycubic. if i can have an help i can be able to upgrade it . 

### Describe alternatives you've considered

i maintain the old version of cura  to let the extenzion work but i should like to migrate

### Affected users and/or printers

all people have a anycubic i3 and similar printers

### Additional information & file uploads

[BedlevelPlugin.zip](https://github.com/Ultimaker/Cura/files/7970512/BedlevelPlugin.zip)

tmirante, 2022-02-02:
quote

fvrmr, 2022-02-04:
Hi @tmirante thank you for your question. 
But we can't help you with upgrading an extension for a third party printer. We only maintain Ultimaker printers and third party printers are contributed and maintained by the community.

closed 2022-02-04 by fvrmr