Ultimaker/Cura #11391: Holes disappear when sliced

scfell, 2022-01-31:
### Application Version

4.13.0

### Platform

Windows 11

### Printer

Cr10s5

### Reproduction steps

Load STL, sliced STL, preview STL, saw missing holes

### Actual results

Holes are missing

### Expected results

See holes

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]
Preview

scfell, 2022-01-31:
This is my 1st bug report, if you need anything please let me know, I was able to slice the same STL on Cura 4.11.0

GregValiant, 2022-01-31:
Go to the Mesh Fixes and turn off "Remove all Holes".  That and "Make Overhangs Printable" are specialty settings for specific problems.