Ultimaker/Cura #11392: Cura 4.13 slicer slower than 4.8

AKO22, 2022-01-31:
### Application Version

Cura 4.13

### Platform

windows 10

### Printer

Artillery Sidewinder 2 x

### Reproduction steps

Ran 4.13 to slice a STL file and took a long time to slice as well as estimated more time to print than 4.8 slicer

### Actual results

longer time to slice and estimated time to print

### Expected results

I would think the updated version of Cura 4.13 would be better slicing time and print time

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Cannot drop file in

fvrmr, 2022-02-01:
Hi @AKO22 thank you for your report.
Could you share your project file (file - save project) in a .zip here on GitHub. 
Thank you in advance.

closed 2022-02-15 by no-response[bot]

Ghostkeeper, 2022-02-15:
We make trade-offs between slicing speed, printing speed, print quality and reliability all the time. Some changes we make cause the slicer to become slower or the print to take longer, but in turn the slice should become more reliable, bugs are fixed or the quality is better.