Ultimaker/Cura #11395: Missing connection to cloud. (4.13.1) (Looks like problems in 2021-11)

TonHuijs, 2022-01-31:
### Application Version

4.13.0

### Platform

windows 10

### Printer

Ultimaker S5

### Reproduction steps

Printer offline sinds few days. Can't connect.
Restart pc and printer (several times) doesn't change the situation.
Update to 4.13.1, no changes in behavior / cloud connection still not working.

### Actual results

Can't reach the s5 printer in the cloud.

### Expected results

The situation looks like the hughes breakdown fuw month ago.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Changing / update to 4.13.1, No additional files added.

fvrmr, 2022-02-03:
Hi @TonHuijs thank you for your report.
Could you share your log file? You can find your log file here: `C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

Ghostkeeper, 2022-02-15:
Could be a firmware issue? If the printer is offline, there's nothing that Cura could do about it. Ultimaker's website would indicate if the printer is still online.

closed 2022-02-17 by no-response[bot]