Ultimaker/Cura #11400: when i close cura

Haiten76, 2022-02-01:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

wanhao D12 230

### Reproduction steps

when i leave the program 

### Actual results

the program crash 

### Expected results

need to relaunch the application and modification are not save

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Traceback (most recent call last):
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1100, in event
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 482, in event
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Event.py", line 218, in call
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 620, in closeApplication
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\Settings\MachineManager.py", line 317, in setActiveMachine
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 676, in setGlobalContainerStack
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Application.py", line 282, in setGlobalContainerStack
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Signal.py", line 219, in emit
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Signal.py", line 332, in __performEmit
  File "C:\Program Files\Ultimaker Cura 4.13.1\plugins\MKS Plugin042802\MKSOutputDevicePlugin.py", line 136, in reCheckConnections
    Logger.log("d", "GlobalContainerStack change %s" % active_machine.getMetaDataEntry("mks_network_key"))
AttributeError: 'NoneType' object has no attribute 'getMetaDataEntry'

Thread 0x00005d58 (most recent call first):
  File "threading.py", line 306 in wait
  File "zeroconf\__init__.py", line 2534 in wait
  File "zeroconf\__init__.py", line 1715 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00001404 (most recent call first):
  File "zeroconf\__init__.py", line 1369 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x0000262c (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.13.1\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00002900 (most recent call first):
  File "threading.py", line 306 in wait
  File "zeroconf\__init__.py", line 2534 in wait
  File "zeroconf\__init__.py", line 1715 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

[... 152 lines cut ...]
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000010f8 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Current thread 0x0000579c (most recent call first):
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 398 in _logInfoWidget
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 184 in _createDialog
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 88 in __init__
  File "C:\build\Cura\4.13-exe\build\inst\bin\cura_app.py", line 162 in exceptHook
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 396 in exec_
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 877 in run
  File "C:\build\Cura\4.13-exe\build\inst\bin\cura_app.py", line 236 in <module>
  File "Console.py", line 36 in run
  File "__startup__.py", line 74 in run

fieldOfView, 2022-02-01:
This seems to be a bug in the [MKSOutputDevicePlugin](https://github.com/Jeredian/mks-wifi-plugin). Please report this issue in their issue queue.

Haiten76, 2022-02-01:
yes you have true, i've launch an update of the plugin and now it work fine, thanks for the help !!

closed 2022-02-01 by Haiten76