Ultimaker/Cura #11401: Double lines on the even numbered layers

GregValiant, 2022-02-01:
### Application Version

4.13.1

### Platform

Windows 10 Home (in compatibility mode)

### Printer

AnyCubic i3 Mega X

### Reproduction steps

This report comes from problem that a poster on the Ultimaker Forum had.  He passed along a project file and a Gcode file.  I was unable to reproduce the problem on my Win 10 computer.  I looked at the gcode in a preview and I read the gcode into AutoCad.  The problem is there.

Monotonic was enabled and ArcWelder was enabled (there are no G2 or G3 lines in the Gcode) and Relative Extrusion was enabled.  There are 6 bottom layers and 6 top layers.  The model is a cube with 100% infill.

When the skin goes down on the even numbered layers (0, 2, 4 etc.) the extrusions start at the lower right corner, the extrusions go down correctly until they get to the middle of the square, but instead of continuing to the far corner, the nozzle returns to the start corner, and from there continues correctly to the far corner.  The result is that 1/2 of the skin on the layer gets double-dipped.
The odd layers, the infill, and the walls are fine.  It's the top and bottom skins that have the problem.

### Actual results

The lines in magenta are double lines.  Layers 194, 196, 198 also show the double lines.  Going through a preview line-by-line on these layers also shows the problem.
I've asked the poster to provide information about his computer system.

Here are a couple of lines from the Gcode from Layer:0.
503   G1 X159.03 Y158.408 E0.23009
504   G1 X159.03 Y158.719 E0.0029
---
622   G1 X159.03 Y158.408 E0.23009
623   G1 X159.03 Y158.719 E0.0029
[image]

### Expected results

One set of extrusions per layer.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Here are the problem Gcode and a project file that the poster provided.  I could not duplicate the problem with that project file.
[Cube100_bug_possible.zip](https://github.com/Ultimaker/Cura/files/7982306/Cube100_bug_possible.zip)
Here is the link to the thread on the [Ultimaker Forum](https://community.ultimaker.com/topic/39206-ultimaker-cura-413/page/2/#comment-302496)

I know it's not much to go on.  Good luck.

Nikolay-Po, 2022-02-02:
Yes, I was original poster of a problem.
I'm working under Debian 10 Linux as a host OS, Intel(R) Xeon(R) CPU E5-2678 v3, 64GB RAM. The bug was noticed in slicing result of "Ultimaker_Cura-4.13.1.AppImage" executable file which I have downloaded some days before by Ultimaker site link.
I created defective G-code twice from same run of an application. But was unable repeat the bug another time, while started AppImage a day later.
At first time, the bug depend on Monotonic Top/Bottom checkbox and when I unchecked this checkbox, the bug was gone.
Also, may be it is relevant, before switching to ZigZag Top/Bottom pattern, I tested Concentric pattern with "Connect Top/Bottom Polygon" option.

fvrmr, 2022-02-03:
Hi @GregValiant and @Nikolay-Po thank you for your report.
I think this is related to https://github.com/Ultimaker/Cura/discussions/10931#discussioncomment-1886436 If you change the top/bottom pattern to lines it shouldn't happen. 
We have a ticket on our backlog for this, I will link this issue to our ticket as well.

GregValiant, 2022-02-03:
Thanks Fenne.

fvrmr, 2022-03-17:
Hi @Nikolay-Po could you share a project file where you have this behaviour happening? I can't reproduce this in 4.13.1 only the in Arachne beta 2.

closed 2022-03-31 by no-response[bot]