Ultimaker/Cura #11402: Cura Crashes on Startup ~80% of the time with an access violation

adobo76, 2022-02-02:
### Application Version

4.13.1

### Platform

Windows 11

### Printer

Ender 3

### Reproduction steps

Run the application.  This issue happens on startup at least 80% of the time.

### Actual results

Cura starts up.  I briefly see the window and it shuts down.

### Expected results

Cura should not crash

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[stderr.log](https://github.com/Ultimaker/Cura/files/7982684/stderr.log)

fvrmr, 2022-02-04:
Hi @adobo76 thank you for your report.
Your log file doesn't show what is causing the crash. Also good the know that we officially don't support Win11 because of our Qt version. I will leave this issue open and hope for other users to have maybe more information about it, since we can't reproduce this ourselves.

adobo76, 2022-02-04:
I found a solution.  I uninstalled the display adapter via device manager.  Then uninstalled the Radeon software.  After reinstalling everything the issue has gone away.  Somehow it looks like something between the windows env and the driver got out of sync.  Note that this issue was happening even though I was already at the latest up to date driver according to the Radeon software.

closed 2022-02-16 by Ghostkeeper