Ultimaker/Cura #11403: Error after print finish

marcok625, 2022-02-02:
### Application Version

4.13.1

### Platform

Windows 10 

### Printer

Ultimaker S5 Pro bundle

### Reproduction steps

1. finish print
2. something go wrong 
3. the bed go up and the printer head go to home 
4. the printer head break the printer part
5. 

### Actual results

the printer say print finished .

### Expected results

I think that the bed go down, not go up. 
This part was already printed, many time. But after update cura there is this problem.
Here videos and photo of the last two print. [https://drive.google.com/drive/folders/1HAK82msya0MrXRic2hvH_QMNL58Ckp6S?usp=sharing](url)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[UMS5_Tray-20-12-2021.zip](https://github.com/Ultimaker/Cura/files/7988490/UMS5_Tray-20-12-2021.zip)

fvrmr, 2022-02-03:
Hi @marcok625 thank you for reaching out. 
Did you use the same project file in 4.13.1? Or did you create a new file with new settings? 
And did you see it happening?

fvrmr, 2022-02-04:
Hi @marcok625 thank you! Let me know if you have an update about it!

marcok625, 2022-02-07:
Hi,

After update the firmware of printer from 7.0.1 to 7.0.2 the printer have finished the print correctly.

Thanks

Marco

fvrmr, 2022-02-08:
Thank you for your reply and good to hear that the printer has printed correctly.
I will close this issue, but feel free to reach out when needed!

closed 2022-02-08 by fvrmr