Ultimaker/Cura #11407: Tree Support

Serafino946, 2022-02-03:
### Application Version

4.13.1

### Platform

Windows 11

### Printer

Anycubic I3 Mega

### Reproduction steps

Generating supports with Tree option  in "Support Structure" and "Touching Buildplate " in "Support Placement", 
generates some supports also in some non Buildplate holes.

### Actual results

--

### Expected results

---

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[AI3M_Drill Bracket-34mm.zip](https://github.com/Ultimaker/Cura/files/7993694/AI3M_Drill.Bracket-34mm.zip)
[Screenshot (1).zip](https://github.com/Ultimaker/Cura/files/7993700/Screenshot.1.zip)

fvrmr, 2022-02-04:
Hi @Serafino946 thank you for your report.
You can change the Tree Support Collision Resolution to 0.1mm when this is changed you will see no support. 
Read here more about this setting: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/support/support_tree_collision_resolution.md

Serafino946, 2022-02-04:
Thanks so much,
i tried with 0.1 and it works.
The 0.2 setting was the default, so I didn't bother to change it.
Very kind of you, for your prompt reply.