Ultimaker/Cura #11408: Many walls vase mode

karabas2011, 2022-02-03:
### Is your feature request related to a problem?

I print on pellet extruder with big nozzle (3mm currently ) with no retract. 
Most my models are vase mode capable and it is preferable printing method. No travel movements at all, no nozzle leaking. No seams.
But it limited to one wall thickness.

### Describe the solution you'd like

 I need a  mode when all walls  in layer become one closed loop. Then  it can be printed in vase mode.
Suboption is rounding semiloop "ends" to more smooth movement

### Describe alternatives you've considered

common way is currently used.

### Affected users and/or printers

Any user can use vase mode on any nozzle diameter and no wall thickness limitation

karabas2011, 2022-02-04:
[image]

fvrmr, 2022-02-07:
Hi @karabas2011 thank you for your feature request.
I will bring this up with the team. Keep you posted!

karabas2011, 2022-02-08:
Preliminary tests show problem with "seam". We need some options for overlapping "seam" ends. Currently no noticeble advantages.

casperlamboo, 2022-02-11:
Hi @karabas2011, This ticket has already been deferred but I thought it was really interesting. Your example of 3 walls can be extended to a general solution that work for a vase mode that work on vase mode with a wall count of any number. In this solution there is a slight difference between even and odd number walls. The image below can be trivially extended to any number walls.
[image]

There is still one problem though: the 2nd 3rd etc walls must be offsetted from the original polygon shape. A offset of a continuous polygon doesn't always end up as a continuous polygon (see attached image below). For vase mode this is a problem as it requires all polygons to be continious, otherwise funky stuff will happen. 
[image]

karabas2011, 2022-02-17:
Yes this is limitation. You need to model part with such slicing in mind. But I faced with more important seam problem. There are no any possibility to overlap seam walls even with no-vase mode. Big nozzle tends to drag extruded plastic line when starts to travel and  creates seam holes.

Ghostkeeper, 2022-02-17:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/4323 . There has been some analysis in the other ticket as well about how this could be approached, and the limitations with what shapes can be printed, water tightness, etc.

closed 2022-02-17 by Ghostkeeper