Ultimaker/Cura #11410: Tree support unnecessarily aggressive branch angles, skipping overhangs, influenced by part rotation or distant parts on plate

tastyratz, 2022-02-03:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Monoprice Maker ultimate

### Reproduction steps

Using tree supports touching the buildplate I was unable to get inner structure support without increasing branch angle unnecessarily over 40 degrees. 

Rotating the supported part on the buildplate influences if a second or third buildplate touch happened at all, even when no other parts are close. 

If I add another part on the plate that's nowhere near the part in question it changes what tree supports are or are not even generated.

This file demonstrates the encountered problem well aligned to widest flat against buildplate.
https://www.thingiverse.com/thing:4051105

### Actual results

Tree support starts low with aggressive early overhang/branch angle and straighter top structure

Structure should not be influenced by part rotation indicating inconsistent calculations

Structure preference for early aggressive overhang/branch angles results in missed structure support

Original plate when I started with a number of misc things to print
[image]
Notice the unsupported overhangs. This is set to 40 deg branch angles.

After setting to 48 deg (and deleting some parts nowhere near the part in question).
[image]

The same part with only 1 other part on plate, far away:
[image]

After deleting second part and reslicing with NO other changes:
[image]

### Expected results

Tree support should average and prefer lower branch angles whenever possible to reduce overhangs and make straighter support lines on average. It does not always need to go wide, then straighter up - when it can prefer more gradual angles.

Tree support branch angle should also operate as a maximum clamp, not a cutoff. Lower branch angle settings should generate tree supports with lower angles if it's possible to do so. We shouldn't be creating overhangs to support overhangs when possible.

This also reduces cantilevered force that tensions against buildplate attachment.

Even though it missed support of top of structure it still angled inward over the part, then came back out. It seems cura still influences early tree support path by overhangs in need of support... which the tree will eventually not support. If it's not going to be supported, the tree path should also ignore an overhang in it's calculations.

Other parts on buildplate nowhere near this one should not change if or how supports are generated at all.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Screenshots and example file linked in bug report to reproduce.

fvrmr, 2022-02-04:
Hi @tastyratz thank you for your report.
Tree support will build up the trees different depended on the models on the buildplate. See https://github.com/Ultimaker/Cura/issues/7797
The missing overhangs seems related to an issue we already have on our backlog. See https://github.com/Ultimaker/Cura/issues/11239  And I will link your issue as an user case to that ticket. Devs see CURA-8840

I will close this ticket as a duplicate.

closed 2022-02-04 by fvrmr

DerGenaue, 2022-02-22:
@tastyratz Dynamically preferring lower branch angles unfortunately is really hard algorithmically as far as I understand it.

As an alternative, you can also check out this alternative Tree Support implementation a friend of mine created to see whether it satisfies your requirements to some degree:
https://github.com/ThomasRahm/CuraEngine

His version does at least one step into that direction by having a preferred and a maximum branch angle