Ultimaker/Cura #11411: unchecked scaling function does not work

Dvrys, 2022-02-03:
### Application Version

4.13.1

### Platform

windows 10

### Printer

ender 3 v2

### Reproduction steps

Loaded an Stl file. 
selected scale function
ensured uniform check box was unchecked
grabbed the x axis to make it longer.

### Actual results

Results: all axis increased in size.

### Expected results

only the axis I select should change.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Using the same steps in Cura 4.11, everything works fine.

fvrmr, 2022-02-04:
Hi @Dvrys thank you for your report. 
I am not able to reproduce this. Could you share a screencast of this happening?

Dvrys, 2022-02-05:
Thanks for getting back to me. 
I wrote these step by step and recorded the symptoms.
The symptoms vary but they are all tied to the scale and rotate functions.

From Thingiverse, Download 
craftsman table saw 137-218250 Mitre slot rail 
by cincyparrothead. Then load into Cura.

Select the scale function and click on the rail model. Ensure snap And Uniform Scaling are checked.

X 15.75 mm     100%
Y154.4 mm.      100%
Z 9 mm.            100%

Grab any axis handle and they all scale uniformly.
Uncheck Uniform and snap scaling. 
Grab any axis handle and only that axis moves..

Select Rotate function and rotate the model to about 45 degrees (so it points at corners).

Select scale again.
Now the settings are as follows

X 120.3142 mm.  100%
Y 120.3142 mm 100%
Z 9 mm 100%

(Sometimes the handles are missing but the values are there and are closer to what they are supposed to be when that happens.)

Note : the model changes from a perfect rectangle to a parallelogram simply by rotating it.

With Uniform and snap unchecked, grab any axis and any combination of axis move at the same time.

I dont have email setup on my pc so it is cumbersome to screen capture then transfer to my Ipad to send pictures. It will take me several  days to send them if I can get them to send at all.
Dan V.

Ill try to take some screen shots if I can.

fvrmr, 2022-02-11:
Hi @Dvrys thank you for your detailed reproduction steps. Unfortunately I still can't reproduce this:
[image]

Could you share a project file (file - save project)? Maybe then I can reproduce it.

Dvrys, 2022-02-12:
Hello again, here is the project file and some pictures.
 

Pic 1 shows model as imported, pic2 shows abt 45 deg rotation, pic 3 shows pulling y axis handle to extend the model.

Also it seems that when you use the rotate function, the scale values represent the area the model occupies on the build plate rather than the actual model itself. Even in your photo,  the scale’s numerical X and Y values indicate a square rather the the rectangular shape of the part. Please Correct me if Im wrong but shouldn't those values be the length and width of the part regardless of how the part is oriented on the build plate?

Dan V.
Dvrys

fvrmr, 2022-02-14:
I can't see your files because you replied via email, you need to add them in the thread [here](https://github.com/Ultimaker/Cura/issues/11411) on GitHub.

Dvrys, 2022-02-14:
Good day, here are the three pictures. Im having problems uploading the project file because Github says it doesnt support  .3mf file format. Ill try to get it to you when I have access to my laptop again. I hope the pictures help though.
Dan V.

[image]
[image]
[image]

fvrmr, 2022-02-14:
Thank you for your photo's. It looks to me like expected behaviour. I don't see the other axis adjusted in your photo's. 
After the rotation you did scale only the y-axis of your object right?

fvrmr, 2022-02-25:
@Dvrys when you rotated your model and the want to scale your model from that position, Cura will scale from that position. So it will not look at it original model for scaling. 
Cura is not a CAD program, so if you want to make sure that your models are designed well, it is better to use software that is for 3D modelling.

Dvrys, 2022-02-25:
Ok. Thank you for the explanation.

Ghostkeeper, 2022-03-12:
This is working as intended then.

closed 2022-03-12 by Ghostkeeper