Ultimaker/Cura #11413: Extruder stops at a random interval.

jones3511, 2022-02-04:
### Application Version

4.12.1, 4.13.1

### Platform

Windows 10

### Printer

Ender 3 V2

### Reproduction steps

Printing with a .8 mm nozzle.

1.select it in nozzle size as .8 and petg for material
2.select draft quality and change nothing (model will print fine)
3. Change any parameters (ex. speed or layer height) and the extruder will stop at a random time.  Sometimes its after the first layer, sometimes its 90% done.
4. I took the same model and sliced it with another software and it worked fine.

### Actual results

SO to sum up any changes to the stock draft profile or any custom profiles cause the extruder to randomly stop.  This is only happening on cura as other slicer software work for the same stl.

### Expected results

Extruder will stop.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura13.1_SLICE_edited_profile.zip](https://github.com/Ultimaker/Cura/files/7999148/cura13.1_SLICE_edited_profile.zip)

fvrmr, 2022-02-04:
Hi @jones3511 thank you for your report.
Could you share a project file (file - save project)?
Are you using USB printing to print?

jones3511, 2022-02-25:
> I just saw this string of emails.  I must've just clicked through them.  I
> attached the project file, and I primarily use octoprint, but in order to
> isolate it to cura I use 3 different micro SD cards directly into 3
> different ender 3 and still had the issue.
>
> Recap: slice and it starts the print just fine and then the extruder
> stepper just stops, with everything else working fine

Should also mention that I have sliced and printed this with Prusa without
any issues.

Ghostkeeper, 2022-03-07:
> I attached the project file

Attachments to e-mail replies don't get posted on Github. You'd have to [go back to the Github website](https://github.com/Ultimaker/Cura/issues/11413) and post the project file there. Although I don't think we can really help you, because...

The g-code you posted in the original post shows simply that Cura instructs the printer to keep extruding material during the print. It doesn't stop at a certain point. So the issue is not with the g-code, but with the printing process. A possibility is that your nozzle is clogging, or perhaps your feeder is slipping. The Settings Guide has [an article](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/unfinished_print.md) about this problem that gives a couple of knobs you can turn to reduce this problem.

If the issue doesn't happen in a different slicer, that just means that the different slicer produced different g-code that doesn't strain your printer as much. You could try emulating the behaviour of PrusaSlicer in Cura. Almost all of PrusaSlicer's features are also possible in Cura.

We can't really help with print process issues. We don't tune the profiles for your printer. We don't even have your printer, and if we did we wouldn't be able to tune profiles for all printers on the market. So we rely on the community to maintain profiles for their printers. If you find a set of settings that works better, feel free to post them to use via a pull request!

fvrmr, 2022-03-17:
Hi @jones3511 you need to attach the project file (as a .zip) in a comment on Github. When you reply via email it won't attach the file.