Ultimaker/Cura #11419: Pause at height

billmail, 2022-02-04:
### Is your feature request related to a problem?

When I have a specific print that requires me to add a "Pause at Height" for a filament change, I would like to see an option to only have "Pause at Height" active for the specific print I want it on, instead of always active once you use this feature.

Too many times, I have started a new print of a different 3D model only to find that the "Pause at Height" is still active and thus it stops the print at the level of the previous 3D print.

I realize this is my fault for not remembering, but it would be nice to have the option to have the software "not" remember this setting when I finish the current print so that I don't inadvertently have it active when I'm on a different 3D model print.
Or at least have an option to save it for future prints or not.

### Describe the solution you'd like

See above

### Describe alternatives you've considered

when starting a new 3D print, have a warning pop up asking if I intended to have "Pause at Height" active.

### Affected users and/or printers

anyone

GregValiant, 2022-02-05:
When there are post-processors active then to the left of the slice button is an icon with a red circle.  The red circle has a number inside indicating the number of post-processors you have active.  Hovering the mouse button over the icon will show a list of the post-processors you have active.  Clicking on the button will bring up the regular post-processing dialog box.  From there you can close out any of them that you don't want active by clicking on the X at the right side.

fvrmr, 2022-02-07:
Hi @billmail thank you for your request.
Could you tell me why the red dot would not work for you.

[image]

billmail, 2022-02-07:
The red dot works if I remember to notice that it is there when I create a
model print.
As I mentioned, this is my fault for not remembering or noticing the red
dot. I just thought it would be easier, with less to have to keep up with,
if there were an option to always turn off any changes from the normal
default settings so that it is only for the print that I set it up for.
I doubt that anyone would want Pause and Height to be on for any other
print unless specifically desired especially since the height would not be
the same for each print.
By the way, I have made changes that show up in the dot, but those were
changes that I specifically wanted as a default for all prints.  So unless
I tell Cura that this is to be a new default setting, it shouldn't
automatically be added for any print.
Just my thoughts.
SG

fvrmr, 2022-02-10:
Thank you for your reply. I understand, I discussed this with a colleague and we decided this is not something we would to add. Since the red dot is enough for now. 
 
And if you want to clear your settings you could start a new project (CTRL + N, or File - New project). This will not remove your settings from the post processing script. 

I will close this issue, but feel free to reach out!

closed 2022-02-10 by fvrmr