Ultimaker/Cura #11420: Fan speed cannot be an integer!

alfrankgit, 2022-02-05:
### Application Version

4.13.1

### Platform

macOS 12.2

### Printer

Ultimaker 3

### Reproduction steps

Load a 3d object.
In "Cooling" set "Fan speed" to 50%.
Slice.
Save to disk.
Open gCode file with text editor.
Search for "M106".

### Actual results

"M106 S127.5" is found.

### Expected results

Either "M106 S127" or "M106 S128" should be found.

The value that M106 expects is a 1 byte integer, not a float!

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

;START_OF_HEADER
;HEADER_VERSION:0.1
;FLAVOR:Griffin
;GENERATOR.NAME:Cura_SteamEngine
;GENERATOR.VERSION:4.13.1
;GENERATOR.BUILD_DATE:2022-01-28
;TARGET_MACHINE.NAME:Ultimaker 3
;EXTRUDER_TRAIN.0.INITIAL_TEMPERATURE:100
;EXTRUDER_TRAIN.0.MATERIAL.VOLUME_USED:1110
;EXTRUDER_TRAIN.0.MATERIAL.GUID:506c9f0d-e3aa-4bd4-b2d2-23e2425b1aa9
;EXTRUDER_TRAIN.0.NOZZLE.DIAMETER:0.4
;EXTRUDER_TRAIN.0.NOZZLE.NAME:AA 0.4
;EXTRUDER_TRAIN.1.INITIAL_TEMPERATURE:205
;EXTRUDER_TRAIN.1.MATERIAL.VOLUME_USED:14
;EXTRUDER_TRAIN.1.MATERIAL.GUID:506c9f0d-e3aa-4bd4-b2d2-23e2425b1aa9
;EXTRUDER_TRAIN.1.NOZZLE.DIAMETER:0.4
;EXTRUDER_TRAIN.1.NOZZLE.NAME:AA 0.4
;BUILD_PLATE.TYPE:glass
;BUILD_PLATE.INITIAL_TEMPERATURE:60
;PRINT.TIME:517
;PRINT.GROUPS:1
;PRINT.SIZE.MIN.X:9
;PRINT.SIZE.MIN.Y:6
;PRINT.SIZE.MIN.Z:0.25
;PRINT.SIZE.MAX.X:213
;PRINT.SIZE.MAX.Y:189
;PRINT.SIZE.MAX.Z:4
;END_OF_HEADER
;Generated with Cura_SteamEngine 4.13.1
T1
M82 ;absolute extrusion mode

G92 E0
M109 S205
G0 F15000 X204 Y6 Z2
G280
G1 F1500 E-6.5
;LAYER_COUNT:5
;LAYER:0
M107
M104 T0 S205
M204 S625
M205 X30 Y30
G1 F600 Z2.25
G0 F4285.7 X115.535 Y90.482 Z2.25
M204 S500
M205 X20 Y20

...

;LAYER:3
M106 S127.5

GregValiant, 2022-02-05:
Maybe your firmware is different?  I've never had a problem with the decimal place(s) with Marlin.
Where in particular did you see _"The value that M106 expects is a 1 byte integer, not a float!"_.  I see that the examples are integers, but I don't see anything that says the S parameter "must" be an integer.
I would say that the number seems to be truncated in the firmware.  If I send M106 S178.5 (70%) the printer displays 69% and if I send M106 S179 the printer displays 70%.

alfrankgit, 2022-02-05:
I've never seen a float used as a value for an Arduino PWM output.

fvrmr, 2022-02-11:
Hi @alfrankgit thank you for your report. 
I tested this on my Ultimaker 3 and it is working for me when changing the fan speed to 50%. It shows in the gcode indeed M106 S127.5 but that doesn't mean it won't work. 
What firmware version are you using?

GregValiant, 2022-02-11:
@alfrankgit - Your title for this bug report is "Fan speed cannot be an integer!"
Then you say "I've never seen a float used as a value for an Arduino PWM output."
If it cannot be an integer, and you've never seen a float used, what do we have left?  What have you seen that is not an integer and not a float?
Marlin does not state that it **must** be an integer.  
@fvrmr says M106 S127.5 works on her UM3.
M106 S127.5 also works on my Ender 3 Pro.

Is your UM3 ignoring the command and the fan is not turning on at the correct speed?

alfrankgit, 2022-02-11:
OK, it should be an integer, not a float.
Every other slicer sets the value as an integer.

fvrmr, 2022-02-14:
But is it not working for you?
Like @GregValiant also mentioned Marlin does not state that it must be an integer. Just a value between 0 - 255.

alfrankgit, 2022-02-14:
It does work, but Cura is still wrong to create the value as a float.
No other slicer I ever used does that.
I didn’t analyze the marlin source code what is expected there.
But a value that has a range from 0 to 255 has no reason to be anything other than a 1 byte integer.
Arduino defines it as that for the pwm output which is what is used on the ATMega 8bit printer boards.
https://www.arduino.cc/reference/en/language/functions/analog-io/analogwrite/

nallath, 2022-02-14:
But the printer is accepting the command, right? If so, what is actually the problem here? The fact that it's theoretically a bit weird?

If the printer accepts the command, how is it wrong? There are quite a few behaviors that any slicer has that no other slicer has, but I wouldn't call that wrong because of that.

alfrankgit, 2022-03-19:
Actually, it does cause a problem.
With Flashforge printers, currently the Adventurer 4.
The printer displays the fan speed (M106 S127.5) as 0%. 
The fan is running but how do I know how fast?

alfrankgit, 2022-03-19:
But this just proves my case: you cannot set a register that is supposed to be a 1-byte integer with a floating point value and hope that it gets through properly.

Once again: the origin of our printers were ATMega based boards, and those use a 1-byte integer for the pwm output.

I cite from the data sheet available below: "– Four 8-bit PWM Channels".

https://ww1.microchip.com/downloads/en/devicedoc/atmel-2549-8-bit-avr-microcontroller-atmega640-1280-1281-2560-2561_datasheet.pdf

alfrankgit, 2022-03-19:
Just look at it from the other side:

What harm would be done if you change the fan speed output of Cura to be a one-byte integer?

Ghostkeeper, 2022-04-10:
So then the firmware properly parses this number, realises that it's a floating point number, encodes it differently and sends the floating point number to the PWM? You'd expect that with C applications like Marlin the output would be encoded the same regardless of the input.

If the firmware sends a float to the PWM output, then it is not doing that according to the spec of that ATMega-based board, apparently. Cura's job is to stay within the specifications of g-code, so far as they are standardised at least. The firmware's job is to translate into signals that your specific printer needs, whether that is 1-byte int, 8-byte float, analogue voltage or whatever it takes to turn the fan. Cura doesn't know what the specs are of the ATMega-based board and doesn't care; this is in the scope of the firmware.

jr42, 2023-01-29:
I can con confirm that this breaks fan control on a Flashforge Adventurer 4: 

Expected result for fan speed at 10%:

M106 S26

With fan control in the LCD at 10% 

Actual result: 

M106 S25.5 

With fan control in the LCD at 0%

GregValiant, 2024-11-16:
The post processor "Advanced Cooling Fan Control" uses all integers for the fan speed (unless "Fan Scale 0-1" is enabled).  Does that address the problem and can this be closed?

closed 2024-12-09 by github-actions[bot]