Ultimaker/Cura #11421: viewport errors

marktransform, 2022-02-05:
### Application Version

4.12.1

### Platform

mac big sur

### Printer

ender 3 pro

### Reproduction steps

Just general use.. its all the time. Maybe started after my mac updated to Big Sur, but cant be sure

### Actual results

The view port loads a bit to the right, dragging it in creates an almpst off centered feel to teh view, neither in perspective or othographic view.
I cannot no click on items, but on select them via the list, but then cannot interact with them to rotate etc, because that involves clicking.
basically unusable now.

### Expected results

blah blah... normal operation

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I dont have any more to say

fvrmr, 2022-02-07:
Hi @marktransform thank you for your report.
Are you using an external monitor? If so see: https://github.com/Ultimaker/Cura/issues/8745
A workaround is to uncheck the option to restore window position on start in the general preferences in Cura.

marktransform, 2022-02-07:
Yes…. three… cura is never on the main screen, it’s a screen thing?
There’s about 100 posts in that thread you sent, mind pointing me to the
solution?
:)
M

or Android

marktransform, 2022-02-07:
OH ! I have one more question.
Do you have an older version of Cura that will run on an old imac (late
2009) running High Sierra?
I cant get it to work. This computer is right next to my printer so Id
rather use that instead of having to walk from my office to my workshop
with a little sd card ;)

Mark

fvrmr, 2022-02-10:
See @fieldOfView comment (and mine as well) for the work around. 
I will close this one as a duplicate. 

For an older Cura release check: https://github.com/Ultimaker/Cura/releases/tag/4.10.0

closed 2022-02-10 by fvrmr

Ghostkeeper, 2022-02-19:
See also https://github.com/Ultimaker/Cura/issues/10575