Ultimaker/Cura #11422: Wrong extruder is written to g-code in a new project

susanrm, 2022-02-06:
### Application Version

4.13.1 and 4.11

### Platform

Mac OS 10.15.7

### Printer

Bibo 2 Touch

### Reproduction steps

If I open Cura and edit a file, then slice it, setting it to E2 (T1), it writes the correct gcode. But if I then do something else, say clear the models from the build plate and slice something new, or open a new project, it no longer sets the extruder correctly. I ensure carefully that the correct extruder is selected and that there are no rafts or supports selected for the other extruder.

I can go into the gcode file and see the error right there; I can also manually edit it, but it's a pain to do each time, and I don't always expect it.

What I have tried:
- create new profiles/delete old ones
- upgrade to 4.13.1
- delete the printer entirely and start over with new profiles

Let me know if you need more info.

### Actual results

Lines 16 to 32, where the incorrect code resides:
**M82 ;absolute extrusion mode
M104 T0 165
M104 T1 165
M109 T0 S200
G21 ;metric values
G90 ;absolute positioning
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z2.0 F400 ;move the platform down 2mm
T0
G92 E0
G28
G1 Y0 F1200 E0
G92 E0
T0
M117 BIBO Printing...**

### Expected results

Correct code, same lines, from the file done before clearing the build plate. 
**M82 ;absolute extrusion mode
M104 T0 165
M104 T1 165
M109 T1 S200
G21 ;metric values
G90 ;absolute positioning
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z2.0 F400 ;move the platform down 2mm
T0
G92 E0
G28
G1 Y0 F1200 E0
G92 E0
T1
M117 BIBO Printing...**

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[problem file.3mf.zip](https://github.com/Ultimaker/Cura/files/8009278/problem.file.3mf.zip)

fvrmr, 2022-02-07:
Hi @susanrm thank you for your report.
I can not reproduce your issue with your project file. And if I clear my buildplate after I loaded your project file, it will automatically go to extruder 1 and changing it to extruder 2 did work for me. 
You could disable the first extruder and then it will automatically pick extruder 2 when loading new models.

susanrm, 2022-02-07:
Regardless of whether you can reproduce it or not, it has been happening to me for a couple of months now. It's very frustrating. Perhaps there is a bad prefs file. If you can tell me what to delete, I can try that. 

It is not isolated to that project file. But my only workaround is to quit Cura after every new project and before starting a new one.

Disabling an extruder is silly when I use both quite a bit. It's a longer workaround than quitting Cura. But the bug exists. Try slicing/saving gcodes from multiple different files in the same session. Eventually the bug should turn up.

fvrmr, 2022-02-10:
I tried several times, opening project files or stl's. But I could still not reproduce your issue and for us it is a bit hard to fix something we can't reproduce. 
Could you maybe share a screencast of what you are doing and that will show the bug? It could be that I have a different workflow than you.

fvrmr, 2022-02-11:
Yes I used your project file and sliced and saved the gcode, after that I opened multiple objects with your profile settings and sliced and saved them.

Ghostkeeper, 2022-02-19:
The problem also occurs for me with that project file. It stems from this line in the start g-code: ```M109 T{initial_extruder_nr} S{material_print_temperature_layer_0, initial_extruder_nr}```

The initial extruder is T0 because the adhesion extruder is that. Problem though is that the adhesion is set to None, so it's not actually going to print anything with that extruder. We should probably correct for that case. Both the front-end (as in the start g-code) and the back-end (for the T1 at the end of the snippet) have it wrong here.

susanrm, 2022-02-19:
Thank you so much for identifying the issue! 😊

x40-Community, 2022-08-17:
I can confirm the bug with Cura 5.1.0 for Linux. A solution would be great.