Ultimaker/Cura #11423: 4.11.0 не удается разогнать принтер до нужных скоростей

Danskiy1, 2022-02-06:
### Application Version

4.11.0 

### Platform

Windows 10

### Printer

onix corexy

### Reproduction steps

с нагревательным блоком Vulcano и подающим механизмом  Titan, принтер, методом подбора скорости печати, соплом 1мм печатает периметр стенки примерно 60 мм\с .  слоем 0.3 мм . Прошивка Lerdge 
 Эти параметры я пытался выставить в конфигурации Cura "скорости печати"

### Actual results

Как я понял, когда указываешь в параметрах принтера диаметр сопла, то автоматически подтягивается приблизительные настройки к этому соплу, и как я понял, что есть в самой программе "защита от идиотов" и некоторые параметры при заданных настройках принтера  имеют свое ограничение... так вот... в конфигурации скорости печати стенок, при указании что сопло 1 мм, принтер больше 40 мм\с  не печатает  , но в остальных конфигурация скоростей, печать дна/крышки, перемещение и т. д . ограничение нет, по крайней мере до нужных мне скоростей хватало 

### Expected results

В итоге... какую бы не ставил скорость печати стенок, больше 40 мм\с принтер не печатает, а остальные скорости печатают вроде как надо... получается ... что для того что бы мне хотя бы достичь 60 мм\с скорости печати стенок, нужно на самом принтере увеличить скорость печати, но.... когда увеличиваешь на самом принтере скорость печати, в тех настроек, к которых нет ограничений соответственно тоже увеличиваются ...  и получается что бы достичь хоть какого то результата , скорость на самом принтере нужно выставлять 400%  а остальные настройки нужно чуть ли не минимальные ставить, что бы на 400% скорости принтера, они соответствовали изначальной нужной скорости. Плюс ко всему с такими параметрами некорректно начинают работать настройки ускорений и рывков... 

Если кто то откликнется, и не посчитает меня "идиотом"  и поверит во все высше написанное, то готов прислать все нужные видео, фото.  и так далее ... и поскольку принтер находится не дома то сейчас нет возможности приложить видео и фото. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

пока что нет файлов, фото и видео ...

Danskiy1, 2022-02-06:
О Боже ! что автоматическое исправление текста сделала с моим постом! тут половину слов заменили )

Nikolay-Po, 2022-02-06:
Версия 4.11, как я понимаю, не актуальна. Пробуйте 4.13.х.
Довольно смело заявлять о баге, не указав, что конкретно вы уже проверили, относящееся к скорости. Пробовали включать отображение всех параметров?
Заявляя о баге, желателно прикрепить сохранённый (небольшой, демонстрирующий проблему) проект, а так же проблемный файл G-кода.
Рассматривали предварительный просмотр в режиме отображения скоростей? Приложите снимок экрана, где на пред.просмотре видна пониженная скорость.

Danskiy1, 2022-02-06:
@Nikolay-Po спасибо что отозвались) ну я не волшебник, только учусь, в дальнейшем постараюсь выдержать все условия публикации багов ) 
[image]

https://user-images.githubusercontent.com/99105593/152699274-6f70ac19-d3e2-4960-ade8-693792c171e7.mp4

вот настройки и видео при 04 сопле

Danskiy1, 2022-02-06:
[image]

https://user-images.githubusercontent.com/99105593/152699376-e53ada91-73e4-4f74-adda-2da97d979039.mp4

вот все тоже самое, только указано что сопло 1 мм 
разница в скоростях очевидна... и про том что это ограничение действует на "скорость печати стенок"

closed 2022-02-06 by Danskiy1

Nikolay-Po, 2022-02-06:
На втором видео не видно, чтобы принтер печатал что-нибудь быстрее этого. Не исключаю, что срабатывает ограничение по объёмной скорости потока.
Сохраните проект Куры (расширение .3mf), сохраните файл G-кода. Приложите.
Может, будет понятно, что ограничивает - слайсер или прошивка принтера. Судя потому, что увеличение скорости в меню принтера работает, похоже на слайсер.

Не пробовали более свежую версию Cura?

Danskiy1, 2022-02-06:
@Nikolay-Po вот для сравнение 2 видео ) 
Спасибо, завтра обновлю до 4.13 и если тоже самое будет, то сделаю проект и g - код 

https://user-images.githubusercontent.com/99105593/152702107-d96c6eb5-353a-4d32-bcf6-2654a0974ab0.mp4

Danskiy1, 2022-02-07:
@Nikolay-Po спасибо, обновил версию до 4.13 Все баги пропали, печатает стабильно

Nikolay-Po, 2022-02-07:
Поздравляю. С почином! Мои первые публичные багрепорты (десяток лет назад или больше) были похожи на ваш :)
Учите английский язык. Он удобен для интернационального взаимодействия.

Danskiy1, 2022-02-07:
@Nikolay-Po   спасибо ) не думаю что что прямо так буду углубляться в знание английского, для того что бы написать "отчет о ошибке"   ))) я написал этот пост, по скольку посчитал нужным добавить свой вклад , что бы программа и дальше развивалась,  но не все так просто,  как оказалось ))

Ghostkeeper, 2022-02-19:
If you expect any help from the majority of the community, you'll have to use English!