Ultimaker/Cura #11424: [4.13.1]

printingotb, 2022-02-06:
### Application Version

4.13

### Platform

win10

### Printer

Custom fff

### Reproduction steps

Enable infill walls.
Turn on fill gaps between walls
See attached project. Settings profile was recently imported from Cura 4.3 where it worked as expected.

Looks related to #8626

### Actual results

Gapfill is printed between inner wall and infill wall

### Expected results

It should not do that.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]
[image]
[Gapfil.3mf.zip](https://github.com/Ultimaker/Cura/files/8009952/Gapfil.3mf.zip)

fvrmr, 2022-02-07:
Hi @printingotb thank you for your report.
I could not reproduce your bug with your project file in 4.13.1. Does this happen with every model?
[image]

fvrmr, 2022-02-11:
Ah I missed that indeed, you are right that this is again a bug in 4.13.1 but since this is not a problem in Arachne and 5.0 I have to defer this, we will not fix this for the current engine. I hope you understand!

printingotb, 2022-02-11:
I understand, was this bug in any of the earlier versions? I may need to go back until l5.0 is released.

fvrmr, 2022-02-11:
I checked 4.12.1 and there the bug was also present. In 4.11 the bug is not present (and 4.8).