Ultimaker/Cura #11425: taking ages to slice

Mk1Luke, 2022-02-06:
### Application Version

4.13.1

### Platform

window 10

### Printer

ender 3

### Reproduction steps

updated to new cura and started to slice at took ages to slice 

### Actual results

took about 10mins to slice a project that only takes 5mins to print 

### Expected results

i would of expected it to slice the project faster if its only a little part

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

i have no files to submit just want the slice to be faster like it used to

fvrmr, 2022-02-07:
Hi @Mk1Luke thank you for your report. 
Did it only happen with one project file? And did this happen in earlier versions?
Could you share your log file? You can find your log file here: `C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

Mk1Luke, 2022-02-07:
Hello very sorry but I found out it was my computer and not he slicer! I tried it on my laptop and it worked fine! Sorry! 

Thanks 
Luke

fvrmr, 2022-02-08:
Thank you for your reply. I will close this issue then!

closed 2022-02-08 by fvrmr