Ultimaker/Cura #11426: Print thin walls not working as intended? (4.13.0+)

dtex91, 2022-02-06:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Voron 2.4

### Reproduction steps

Load in a 3Dbenchy model. Whatevery options i check, the results are the bad gaps seen in the screenshot OR bad filled by cura.

### Actual results

I cant get a nice single line from cura, when there is a wall which is less than nozzle dia (0.4mm).
[image]
[image]

### Expected results

These should be a single (thin) line, filling the gap between the walls.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This issue also was present version in 4.13.0

fvrmr, 2022-02-07:
Hi @dtex91 thank you for your report.
Could you share a project file (file - save project) as a .zip here on Github?

dtex91, 2022-02-07:
[GABS_3DBenchy.zip](https://github.com/Ultimaker/Cura/files/8015971/GABS_3DBenchy.zip)

What i have done? Start cura, put benchy in it.

fvrmr, 2022-02-08:
Thank you for your project file. 
You need to change a bit more settings, first of all set `Fill Gaps Between Walls` to `Everywhere` and change the `Top Surface Skin Line Directions` under Experimental from [0] to [ ] (or click on the fx icon, this will restore the calculated value).

4.13.1:
[image]

Or you can use the Arachne engine: https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_beta_2 where you will have variable line width.

Arachne:
[image]

Let me know if that works for you!

dtex91, 2022-02-11:
Yep this worked. But it should be able to do both, [0] degree top surface skin lines and print thin walls, shouldnt it?