Ultimaker/Cura #11428: Customizable amount of fisrt draft shield layers

ANTONIOPSD, 2022-02-07:
### Is your feature request related to a problem?

Cold air goes down, hot air goes up, by making the draft shield layers higher than the main part, we create a small pool that will contain more hot air and cold air will have to displace more hot air so it will heat at the same time it tries to go down, and by the time it reaches the part, that air will be a little bit hotter.

### Describe the solution you'd like

Example:
 1. Start printing
 2. Print 10 0.2mm draft shield layers
 3. Print 1st 0.2mm main layer
 4. Continue with single 0.2 layers everywhere.

Normally draft shield are not very close to the main part so the nozzle won't hit it, of course this could be customizable so we can adjust the values based on the size of the nozzle/hot end.

### Describe alternatives you've considered

One of the alternatives is to first print a full height draft shield with an offset a little bit bigger than the hot end size  so the hot end can't hit ir and the cold air cannot cross that pool of hot air.

### Affected users and/or printers

All

fvrmr, 2022-02-07:
Hi @ANTONIOPSD thank you for your feature request.
I will bring it up with the team, but keep in mind that draft shield is in Experimental so the chances are that we will not pick this up to implement.
Keep you posted.

fvrmr, 2022-02-08:
I have discussed it with the team and we decided to defer this request because it will also be complex to implement this. You also need to take the axis of the printer into account, which are different for a lot of printers. I hope you understand.

closed 2023-06-29 by ANTONIOPSD