Ultimaker/Cura #11429: [4.13.1] Ironing happens in the air. Moving the object moves the issue.

printingotb, 2022-02-07:
### Application Version

4.13.1

### Platform

win10

### Printer

customfff

### Reproduction steps

See attached project and pictures. Ironing is on.

### Actual results

Ironing happens inside a wall and goes outside the model.
Moving the part around moves or eliminates the issue.
Second and third pic is after changing position to Z-1 and the issue is now in a different area.

### Expected results

Ironing only where it's supposed to be, nothing should change by moving the object around.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]
[image]

[iron bug.3mf.zip](https://github.com/Ultimaker/Cura/files/8013939/iron.bug.3mf.zip)

fvrmr, 2022-02-07:
Hi @printingotb thank you for your bug report.
I will bring this up for discussion with the team. Keep you posted!

fvrmr, 2022-02-08:
I have discussed it with the team and since it could not be reproduced in Arachne we decided to defer this issue. I hope you understand!

printingotb, 2022-02-09:
Looking forward to the 5.0 beta!