Ultimaker/Cura #11430: By "Projekt save"  The software always hangs up.

Netcat1967, 2022-02-07:
### Application Version

4.13.1

### Platform

Windows 7

### Printer

Ultimaker 3 Ext

### Reproduction steps

After which I finished my project with arrangement and slicen I want to save the finished project.
The software crashes every time.
I have on a printhead PLA and the other is labeled as empty. 

### Actual results

The software can only be restarted afterwards I can do everything new but a saving as a project does not work.
It still worked for the last version. 

### Expected results

Restart the Software

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8017888/cura.log)

fvrmr, 2022-02-10:
Hi @Netcat1967 thank you for your report. 
Unfortunately we don't support Windows 7 anymore, so that could be the cause of your issue. I don't see anything in the log that could tell my why this is happening. Does it also happens when you want to save a 3mf (not a project file)?

Netcat1967, 2022-02-10:
Hi Fenne,

The problem has only occurred with the last software update.

And to your question, yes I can save an object as 3mf.

 

regards

Volker

fvrmr, 2022-02-11:
Thank you for your reply. Then I am afraid that this is an issue with not supporting Windows 7. 4.10 was the latest Cura version which was supported for Windows 7: https://github.com/Ultimaker/Cura/releases/tag/4.10.0 

I will close this issue since this is something we won't fix unfortunately.

closed 2022-02-11 by fvrmr

Netcat1967, 2022-02-11:
Hallo Fenne,

thanks for your replay.

regrads

Volker

up. (Issue #11430)

<https://github.com/Ultimaker/Cura/issues/11430#issuecomment-1036147977> ,
T3VDANCNFSM5NYJBWVQ> .
t=8&pt=524675>  or Android
tm_campaign%3Dnotification-email%26utm_medium%3Demail%26utm_source%3Dgithub>
<https://github.com/notifications/beacon/AGJ6CRJUILCCJBBMW4HNMDTU2T3VDA5CNFS
M5NYJBWV2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOHXBF2CI

"https://github.com/Ultimaker/Cura/issues/11430#issuecomment-1036147977",
"https://github.com/Ultimaker/Cura/issues/11430#issuecomment-1036147977",
"https://github.com" } } ]