Ultimaker/Cura #11432: Color Scheme: Temperature

c9498, 2022-02-07:
### Is your feature request related to a problem?

Not related to a problem. 

### Describe the solution you'd like

It would be nice to have a graphical representation of hot end temperature in the color scheme dropdown, (like flow). It would be useful for to verify where temperature changes  are occurring particularly when setting up calibration. (Temp towers etc)
Thanks

### Describe alternatives you've considered

None

### Affected users and/or printers

All

fvrmr, 2022-02-10:
Hi @c9498 thank you for your feature request.
I will bring this up with the team. Keep you posted.

fvrmr, 2022-02-11:
I have discussed it with the team and we decided to defer this request. I would be a nice to have though, but it will not get a high priority to implement this. Also temp towers will not work for this, since that is with the post processing script. And post processing scripts are not in visible in the preview, they will be processed while creating the gcode.

closed 2023-07-13 by github-actions[bot]

tripletto, 2023-07-23:
This missing feature is actually the reason, I need to use Prusa Slicer as well. An implementation of a visual temperature preview would be greatly appreciated!

donoit, 2026-08-19:
CURA does not have a temperature color scheme in its preview mode; feature is rejected - thats a pitty !!