Ultimaker/Cura #11436: Can't connect to Ultimaker S5 by hardwired LAN with Cura 4.13.0

BuildingItRight, 2022-02-08:
### Application Version

4.13.0

### Platform

Win 10 Pro

### Printer

Ultimaker S5

### Reproduction steps

Can't connect to the S5 printer running 7.0.2 to transfer files etc. using a hardwired LAN.  Printer can't be seen but has a valid IP.  Problem is present 90% of start-ups.  Multiple reboots and re-starts don't change anything.  Start order is irrelevant.  Problem did occur occasionally on previous software/firmware combinations but far less commonly.  

### Actual results

Printer doesn't show up in configure printer window.  No connection available.  Can't transfer files

### Expected results

Printer should show up in configure printer window.  Printer settings should download to Cura.  Files should be able to be uploaded to printer.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

.

nallath, 2022-02-09:
Did you also try to add the printer IP directly in Cura or did you use the auto discovery? Is the printer available if you go to the IP via the browser?

BuildingItRight, 2022-02-09:
Yes, I can access the printer directly by IP address.

The printer was added manually and shows up in the list of connected printers.  It just won't "connect" and therefore can't be seen to transfer files etc.  Refresh, restarting both the software and printer, rebooting the machine, none of this solves the issue.

I have seen this occasionally in previous versions but a restart or reboot generally fixed the issue.

Thanks

nallath, 2022-02-10:
Could you then also share the log files generated by Cura?

closed 2022-02-24 by no-response[bot]