Ultimaker/Cura #11438: Wall problems (very bad Quality) Cura 4.13

Bastlwastl84, 2022-02-08:
### Application Version

4.13.1

### Platform

win10

### Printer

Voron 350

### Reproduction steps

Nozzle 0.6 and 0.8 (a bit lower with 0.4, but also happens here)

### Actual results

artefacts at rendered previe
[image]
[image]
w

### Expected results

no artefacts

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

also happens with prconfigured basic profiles

GregValiant, 2022-02-09:
The white "artifacts" represent the Z seam location.  On the right side those look like an interference between the inner and outer walls and possibly Cura is showing it that way because the flow has been adjusted.  It could also be a defect in the model, or possibly a display issue.  A project file would likely tell the story.

closed 2022-02-24 by no-response[bot]

Ghostkeeper, 2022-02-26:
The adjusted flow on the right there is caused by the Wall Overlap Compensation setting. It is meant to give a more accurate result, at the cost of more flow changes.