Ultimaker/Cura #11439: Spiralized Outer Contour gcode mistake

kjs-DE, 2022-02-08:
### Application Version

4.13.0, 4.13.1

### Platform

Xubuntu 18.04

### Printer

Anycubic Mega S

### Reproduction steps

generated a block STL with some cut out
sliced with Spiralized Outer Contour 0,4mm nozzle, 0,3mm height, 0.8mm width
[image]

[box-inserts.zip](https://github.com/Ultimaker/Cura/files/8027622/box-inserts.zip)

### Actual results

As can be seen in the above picture one line makes a shortcut thus making it unusable for printing. With simpler shapes it worked just fine but here it failed.

### Expected results

all lines to follow perimeter

(Sorry, first time filing a github report)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

none

smartavionics, 2022-02-08:
See https://github.com/smartavionics/Cura/releases for a Cura variant that gets this right...

[image]

kjs-DE, 2022-02-08:
Thanks! Works like a champ with that version. However, if I increase the notch width from 10 to 12mm 4.13.1 and 13.4.0 will also work like they should.

smartavionics, 2022-02-09:
Good, glad you have a workaround. Sometimes, changing the layer height or line width can also solve weird problems like this.

fvrmr, 2022-02-10:
Hi @kjs-DE thank you for your report.
I could not reproduce it on Windows. But I checked with a colleague who is using Linux and he could reproduce it. So this is a duplicate of https://github.com/Ultimaker/Cura/issues/9979. We have a ticket on our backlog for this. 
I will close this as a duplicate.