Ultimaker/Cura #11440: 4.13.1 AppImage doesn't start on linux

wileu, 2022-02-09:
### Application Version

4.13.1

### Platform

Mint 19

### Printer

TwoTrees Sapphire Pro

### Reproduction steps

Double click on the file

### Actual results

Nothing - no errors, etc.

### Expected results

To run the program.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Something.

fvrmr, 2022-02-10:
Hi @wileu thank you for your report.
What file are your trying to open a stl or you want to open Cura?
Could you share your log file? You can find your log file here: `$HOME/.local/share/cura/<Cura version>/cura.log`

wileu, 2022-02-10:
I want to open Cura. I only have a folder for other Cura like 4.12 and 4.13.0. Cura 4.13.1 does not create any folder, file. Cura 4.13.0 opens without a problem.

wileu, 2022-02-11:
Yes, I know. I write that Cura 4.13.0 create folder 4.13, but 4.13.1 doesn't create any folder like 4.13.

wileu, 2022-02-11:
I already know what's wrong. The problem is the GLIBC. Cura 4.13.1 needs GLIBC 2.78  but I have version 2.27.Regards.

closed 2022-02-11 by wileu

lucastcox, 2022-02-23:
Still an issue. Says it needs glibc 2.28 and I have 2.27

Ghostkeeper, 2022-02-26:
I wonder what component says that then. We didn't implement that message ourselves. Qt mentions it needs GLib 2.8.3. Perhaps it's Sip? Or Python?

GWRon, 2022-03-03:
Doesnt the glibc requirement come simply from the building OS ? 
In other words: building on an older distribution will lead to a lower glibc required.

Maybe cura should not try to run on the latest versions of its dependencies.

Ghostkeeper, 2022-03-12:
Ah, that sounds plausible indeed. If it's linking to the system version of that library it might not be binary compatible with older versions. We used to build on CentOS6, but changed that to Debian-buster with 4.13. It was apparently necessary for the upgrade to Python 3.10 in the upcoming release, which we were already preparing for back then.