Ultimaker/Cura #11451: "No profiles matching configuration of this extruder" error

mrpegman, 2022-02-09:
### Application Version

4.13.1 (Plus older releases)

### Platform

Windows 10

### Printer

Creality Ender-3 Pro but occurs on the CR-10S as well

### Reproduction steps

1. Go to Preferences > Configure Cura > Materials
2. Make a [Duplicate] of the Generic PLA
3. Change the Display Name/Brand/Material Type/Color under [Information]
4. [Close]
5. In the app I can change the material to the new type and all is fine.
6. Go back into the Material configuration and change the Printing Temperature/Build Plate Temperature under [Print Settings]
7. [Close]

### Actual results

There's now a yellow triangle by the material dropdown.  Hovering over it shows "There are no profiles matching the configuration of this extruder".  Just to the right, it says "Unsupported" where it normally shows "Standard", "Dynamic", etc.  There doesn't seem to be any way to get back the quality settings other than never defining any custom materials.

### Expected results

Materials should be completely independent of the extruder except for maybe the Diameter.  Of course, the filament has to fit the nozzle.

If this is the actual way this material vs extruder is supposed to work, provide some more information in the email like, "No extruder can reach X degrees".  That will at least tell me what's wrong and what I can fiddle to get things back right.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Nothing else to add.  Thanks for checking into this.

fvrmr, 2022-02-10:
Hi @mrpegman thank you for your report. 
If you change your Material type to for example PLA R, Cura can't select the right profiles because PLA R is an unknown material type for Cura. And then you will see the warning icon and not supported. Because there are no profiles for that material. It doesn't mean you can't print, but it is more a warning that you need to be aware of. See also: https://github.com/Ultimaker/Cura/issues/8229
We have a ticket on our backlog to add a dropdown menu for material type. So I will close this issue and mark it as a duplicate. Please continue the thread in #8229, so we keep everything in one place.

closed 2022-02-10 by fvrmr

rsilvers129, 2022-05-09:
Change the material type to PLA and that will fix it.