Ultimaker/Cura #11452: GcodeStartEndFormatter with basic math functions

MarcelMo, 2022-02-09:
### Is your feature request related to a problem?

A good start/stop script should be parametric, but since even the simplest scripts need unit conversions (travel speed for example), there is no "one for all" script possible.

### Describe the solution you'd like

GcodeStartEndFormatter should be able to perform at least some basic math functions, and also, should be able to parse the most basic syntax conventions.
My guess, best way would be to define some basic allowed functions and operators.
Functions like sqrt(),pow() and operators like "+","-" and "("/")"
I never used python, so I don't know what kind of evaluation would be safest.

### Describe alternatives you've considered

Alternatively I would suggest allowing the whole script to be python, and generating the complete gcode via script.
Or maybe there is some sort of plugin for that already ?

### Affected users and/or printers

All users would benefit from universal adapting default scripts.

fvrmr, 2022-02-11:
Hi @MarcelMo thank you for your feature request. 
I will bring this up for discussion with the team. Keep you posted.

Ghostkeeper, 2022-02-15:
Possibly a duplicate of https://github.com/Ultimaker/Cura/issues/6701 or https://github.com/Ultimaker/Cura/issues/9906.

MarcelMo, 2022-02-15:
> Possibly a duplicate of #6701 or #9906.

Yes, the error reporting as mentioned should also be adressed. The whole parsing stops on a single error. All other lines should be processed. Maybe the error can simply be added as a comment in the resulting gcode. Or maybe it is even possible to highlight any errors while editing. Also, while editing a suggestion on all possible variables would be really great (maybe even containing the human readable Descriptiong of the variable and/or its source).

fvrmr, 2022-02-18:
We have discussed it with the team and we decided to defer this issue, like the other two. 
It will not get a high priority for us to implement. I hope you understand.

fvrmr, 2022-02-18:
You are free to implement this yourself and create a pull request :) 
We have a backlog full with other more important things to work on and don't expect this to get a high priority on our backlog, but as you can see in the other threads people would like to do this as well.

fieldOfView, 2022-02-18:
I must say that I like the idea of using an existing template engine. In Python, you could use eg [Jinja](https://jinja.palletsprojects.com/en/3.0.x/) or [Mako](https://www.makotemplates.org/) instead of Twig. 

The problem remains though that much of the decisions of eg which extruders are actually used, and which s used as the first extruder, are made in CuraEngine, not in the Python-based frontend that deals with the start and end gcode formatting.

MarcelMo, 2022-02-18:
I search the web a bit.
The coding itself is not the problem. I just don't know about, dependencies, policies, compatibility and configuration options - if needed to switch usage of the engine.
I would need something like jinja2 added to the dependencies and policies, but have no clue how this project handles such a thing. The syntax is slightly different. But usage is simple and straight forward.