Ultimaker/Cura #11454: No more Custom printer, for dual extruder?

DragonFlyer007, 2022-02-09:
### Application Version

4.12

### Platform

Win 10

### Printer

Ender 5 Plus

### Reproduction steps

Machine settings, but not finding custom printer anymore

### Actual results

Not possible to add custom printer, with virtuel extra extruder.

### Expected results

Want possible to add a new virtual extruder.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Want possible to add a new virtual extruder.

GregValiant, 2022-02-10:
When I add a "Custom FFF" printer it can have up to 8 nozzles.  If you want to change that Ender 5+ then you must make changes to the "Creality_Base.def.json" definition file and add extruder definition files to the Extruders folder.

DragonFlyer007, 2022-02-10:
[image]

(Nothing further down either)

nallath, 2022-02-10:
Do you have the startup optimizer enabled? You might have enabled that.

Also, please share the log files.