Ultimaker/Cura #11455: Plugin Development: Improve documentation of Settings API

jeremysalwen, 2022-02-10:
### Is your feature request related to a problem?

The documentation for plugin development is pretty sparse.  Specifically I am struggling with the API for changing settings. 

I am trying to set the Per-Model settings for a given node.  I copied existing code I saw in Cura:

```
container = node.callDecoration("getStack").getContainer(0)
container.setProperty("line_width", "value", "0.1")
```

However, if the setting isn't selected as visible, this has no effect, and the following warning is logged:

```
WARNING - [MainThread] UM.Settings.InstanceContainer.setProperty [372]: Tried to set value of setting when the container has no definition
```

The meaning of all these terms and objects and extremely opaque: container, InstanceContainer, property, definition, stack, etc are all jargon that don't seem to be defined anywhere (or perhaps have multiple definitions?). https://github.com/Ultimaker/Cura/wiki/Profiles-&-Settings gives a very high level overview, but is incomplete and doesn't include any code samples.

### Describe the solution you'd like

Further documentation with code demonstrating how to set the settings at different layers of the stack.

More commented example plugins that perform basic operations would also help.

### Describe alternatives you've considered

Asking on the Ultimaker forums repeatedly.

### Affected users and/or printers

Any plugin developers.

nallath, 2022-02-10:
You're right that we should have better documentation outside of the code to tie some things together! 

But for the moment, you can already find much more information in the code itself: https://github.com/Ultimaker/Uranium/blob/1eacb050882b7ad63dfd1e731ee35eafbd07c93d/UM/Settings/SettingDefinition.py#L69-L88 

I will try to carve out a bit more time to improve the documentation over the next few days.

nallath, 2022-02-11:
I've updated the documentation in https://github.com/Ultimaker/Cura/wiki/Profiles-&-Settings, as well as some changes in the code to make the logging / comments clearer. Looking back at what we wrote down, I can totally see the confusion that it caused with you (and also how we didn't notice, since we were so used to the shorthands that we used)

So does this help you to do what you want to do? 

On a second note; What is is that you're trying to do? Maybe we can also give some other pointers on how to do this better (or update different / more documentation).

nallath, 2022-02-14:
> Also just the basic code snippet of e.g. "here is how we get the Per-Model settings from a node" would be nice.

I'm a bit doubting about that, as it's something that is rather specific; only very few plugins ever that. I might add some extra documentation regarding the types of plugins / explain how current plugins do certain things at a later point.

I have added some schematic overview as to how the per object stacks work to the page, as that is a bit more generic (and also makes sense to explain conceptually). 

As for the code in the plugin, what you do makes sense :) What you could also try is just set the parameters and slice the object multiple times. So instead of generating a single g-code, you generate multiple g-codes that have different settings. This does have the downside of needing intervention of an operator, but it also removes the effect that the position of the model on the buildplate has on the final result (which can be rather dramatic in quite some cases!)

jeremysalwen, 2022-02-14:
Fair enough :D  I suppose that specific example was more relevant to me than most plugin developers, but my main suggestion is that adding a few examples to an explanation makes the explanation much more helpful.  A lot of times I struggle to "ground" what I am reading in something concrete in the codebase or UI.

I like a lot the new diagram you added :)  The documentation is a lot better now, so you could close this issue if you like.

p.s. thanks for the feedback on my plugin!  I did think of printing one example at a time, but the difficulty is that would require like 20x as much manual intervention to work, which sort of defeats the main purpose of the automatically tuning.

closed 2022-03-14 by nallath