Ultimaker/Cura #11458: Incomplete Gcode generation on Windows

MiPnamic, 2022-02-11:
### Application Version

4.13.1

### Platform

Windows 10 Pro

### Printer

Artist-D

### Reproduction steps

- Import some stl, slice it, save file, print.
or
- open an existing project, slice it, save file, print.

### Actual results

Gcode generated by my windows version are incomplete.

See attached Gcodes and compare them.

### Expected results

Same settings on my Macbook didn't lead to the same result.
Even the same project, created on Windows, opened on MacOs, sliced and saved from there will work.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Project_created_in_windows.3mf.zip](https://github.com/Ultimaker/Cura/files/8046304/Project_created_in_windows.3mf.zip)
[Gcodes.zip](https://github.com/Ultimaker/Cura/files/8046309/Gcodes.zip)

fvrmr, 2022-02-11:
Hi @MiPnamic thank you for your report. 
Could you check if your end g-code (in machine settings) on Windows is empty? Or not the same as you have on your Mac?

MiPnamic, 2022-02-12:
Hi @fvrmr.
Checked that, same settings on both.

In the gcodes I attached you can see that the Windows one ends way before the last layer, it should end at Z80.7 but the code stops at Z76.2.

fvrmr, 2022-02-14:
Hmm strange. I can't reproduce your issue. Here is my file that I sliced on Windows 10.
[CFFFP_Umarell_def.zip](https://github.com/Ultimaker/Cura/files/8059557/CFFFP_Umarell_def.zip)
And that looks the same as yours on Mac. Did it only happen with this file or more?

MiPnamic, 2022-02-14:
All files, no matter if the project is created there or on another computer, I had no issues with version 14.11, since upgraded this happened.

I'll try to clean up and do a fresh install of the 14.13 version and retry again.

MiPnamic, 2022-02-17:
Uninstalling and Re-installing the application solved the issue.

Thank you so much, next time I'll try this approach before.

closed 2022-02-17 by MiPnamic