Ultimaker/Cura #11459: top bar menu buggy

zewonex, 2022-02-11:
### Application Version

4.11.0

### Platform

Linux Fedora; Gnome DE; Cura as Flatpak-App

### Printer

Tenlog TL-D3 Pro

### Reproduction steps

Open Cura and click on the menu-bar at the top of the screen (where your preferences, file loader, etc. are)

### Actual results

The selection goes one field left, and when clicking with the left mouse tap it goes one more left further.

### Expected results

I click on a tab and it opens it, where I can select one sub-point

https://user-images.githubusercontent.com/99470494/153560181-c82dfd22-a346-4cba-9445-1968df11698a.mp4

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

On the RPM-Version this issue doesn´t exist, but is unusable for me because I can`t save files

fvrmr, 2022-02-11:
Hi @zewonex thank you for your report. 
Did you also have this behaviour with AppImage? 
You can download the AppImage here: https://ultimaker.com/software/ultimaker-cura

closed 2022-02-25 by no-response[bot]

Ghostkeeper, 2022-02-26:
See also https://github.com/Ultimaker/Cura/issues/10608 .