Ultimaker/Cura #11461: Temperature settings are not adopted

1800Coconut, 2022-02-11:
### Application Version

4.5

### Platform

Windows 7

### Printer

Creality Ender 3

### Reproduction steps

I sliced the project with every setting I did like a many times before.

After trying to print my project, it seems like the temperature settings are not saved.

The bed temperature is set as 0 on my printer and the nozzle temperature too.

I think this problem occured after an update to the new Cura version.

When I try to print one of my old projects, everything is alright.

### Actual results

Temperature is set to 0 when I try to print.

### Expected results

The bed temperature should be 60°C and the nozzle temperature 200°C.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

None.

fvrmr, 2022-02-11:
Hi @1800Coconut thank you for your report.
Could you share your project file (file - save project) as a .zip? Also good to know that we don't support Windows 7 anymore, so did this happen in 4.5 or newer?

1800Coconut, 2022-02-11:
It's the version 4.13.1
I don't know where i saw that it's the 4.5 version
[CE3_Zähler_Einwegverzahnung.zip](https://github.com/Ultimaker/Cura/files/8048299/CE3_Zahler_Einwegverzahnung.zip)
n

fvrmr, 2022-02-11:
Thank you for your reply.
When slicing the project file this is the gcode output:

```
;FLAVOR:Marlin
;TIME:4714
;Filament used: 3.57119m
;Layer height: 0.2
;MINX:82.457
;MINY:85.79
;MINZ:0.2
;MAXX:146.718
;MAXY:150.58
;MAXZ:22
;Generated with Cura_SteamEngine 4.13.1
M140 S60
M105
M190 S60
M104 S205
M105
M109 S205
M82 ;absolute extrusion mode
; Ender 3 Custom Start G-code
G92 E0 ; Reset Extruder
G28 ; Home all axes
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
G92 E0
G92 E0
G1 F2700 E-5
```

M140 S60 is the bed temperature
M104 S205 is the hot end temperature 

And these are the settings in your Cura project:
[image]

So it looks correct to me. Are you printing via SD card or USB?

1800Coconut, 2022-02-11:
Via SD card
And that's the thing I don't get
In Cura everything is set correct, but after I put the SD card in my printer, the temperature is set at 0°C
Nothing is going to heat up, but when I use a file from yesterday (before the Cura update) it's working completely correct

fvrmr, 2022-02-11:
Could you share that g-code file where it will not heat up and the file that works? 
Did you use the same SD card?

GregValiant, 2022-02-11:
Have you tried re-formatting that SD card?  Neither Cura nor Windows can tell if the card has any bad sectors.  Consequently, parts of the gcode file can written to a bad sector.  When the printer receives it, it can't understand those parts and just continues on until it reaches some code that it can understand.  I've had situations where the print starts 1/2 way through the skirt instead of doing the purge lines (but I had manually pre-heated).  One way to check is to open the file that is on the SD card using Windows Notepad.  In my situation, the beginning of the gcode file was computerese gibberish.  There were some cool looking characters in there, but it wasn't gcode text.

1800Coconut, 2022-02-11:
This is the file that worked
[CE3_Gehäuse.zip](https://github.com/Ultimaker/Cura/files/8048847/CE3_Gehause.zip)

And the file that don't work is the one I have uploaded before, the CE3_Zahler_Einwegverzahnung.zip

Yes I used the same SD card, and I deleted and saved it several times because I thought it broke during saving

1800Coconut, 2022-02-11:
So, I formated the SD card and downloaded the file again, still don't work

GregValiant, 2022-02-11:
The printer must be seeing the commands but it isn't responding.  When you pre-heat using the LCD controls does it work correctly?

1800Coconut, 2022-02-11:
> The printer must be seeing the commands because they are in the Gcode, but it isn't responding. When you pre-heat using the LCD controls does it work correctly?

No I have tried that too

1800Coconut, 2022-02-11:
> [CE3_Zahler_Einwegverzahnung.zip](https://github.com/Ultimaker/Cura/files/8049590/CE3_Zahler_Einwegverzahnung.zip)
> 
> Could you try this gcode file? And see if that works? This is the gcode I created from your project file.

Thanks for your effort, I will try that when I'm back home. I will notice you if it worked

GregValiant, 2022-02-11:
If the temperatures don't change using the LCD control then the problem is definitely the printer.  You are going to have to go back to Creality.

1800Coconut, 2022-02-11:
> If the temperatures don't change using the LCD control then the problem is definitely the printer. You are going to have to go back to Creality.

No no, don't misunderstand me. Maybe I have read that wrong. When I choose the file AND turn the heat up by myself, it still doesn't work. I thought he means that

GregValiant, 2022-02-11:
"When I choose the file **""**AND turn the heat up by myself, it still doesn't work**""**. I thought he means that"

I guess I don't understand.

You turn on the printer.  You immediately go to the LCD and go to "Preheat PLA" and "Both" and nothing happens?

GregValiant, 2022-02-12:
_"but i stays at 0°C because the value from the file says so"_
That is not what @fvrmr or I saw from your project file.  The lines tell the bed to heat to 60° and the hot end to heat to 205°.

If the heater and bed will not work from the LCD then it is definitely a printer problem.

1800Coconut, 2022-02-12:
Sry missclick

I understand what you mean

I can control the temperature from the LCD and it's heating, but the file value says 0, so it does not start. And I can't comprehend that the problem is with the printer, because I can print older files without any problem.

GregValiant, 2022-02-12:
Open the bad Gcode file in Windows Notepad.  Look down to the lines following "Generated with Cura...".  You should see the lines as fvrmr posted.  Now open one of the old Gcode files that works.  The same lines should be present.
The lines M104, M109 should be your hot end temperatures for the print.  The lines M140 and M190 are the bed temperatures for the print.  Do any of those lines contain "S0"?

1800Coconut, 2022-02-12:
> [CE3_Zahler_Einwegverzahnung.zip](https://github.com/Ultimaker/Cura/files/8049590/CE3_Zahler_Einwegverzahnung.zip)
> 
> Could you try this gcode file? And see if that works? This is the gcode I created from your project file.

So I tried your file and it works

The printer immediately starts to turn up the temperature, have you changed something?

1800Coconut, 2022-02-12:
> Open the bad Gcode file in Windows Notepad. Look down to the lines following "Generated with Cura...". You should see the lines as fvrmr posted. Now open one of the old Gcode files that works. The same lines should be present (maybe different temperatures). The lines M104, M109 should be your hot end temperatures for the print. The lines M140 and M190 are the bed temperatures for the print. Do any of those lines contain "S0"?

;FLAVOR:Marlin
;TIME:4714
;Filament used: 3.57119m
;Layer height: 0.2
;MINX:82.457
;MINY:85.79
;MINZ:0.2
;MAXX:146.718
;MAXY:150.58
;MAXZ:22
;Generated with Cura_SteamEngine 4.13.1
M140 S60
M105
M190 S60
M104 S205
M105
M109 S205
M82 ;absolute extrusion mode

fvrmr, 2022-02-14:
> So I tried your file and it works
> 
> The printer immediately starts to turn up the temperature, have you changed something?

Good to hear that it worked! Could you let us know what the values are that @GregValiant showed above?

GregValiant, 2022-02-14:
Yes, that happens when you open a Gcode file in Cura.  Cura cannot edit the gcode.

You need to open the 3mf file and then check as I explained.

1800Coconut, 2022-02-14:
[image]

What do I have to open there?

GregValiant, 2022-02-14:
Just open the 3mf file with Cura.  It is the one contained in your original post within the zip file.

1800Coconut, 2022-02-15:
These are the sttings from fvrmrs 3mf file

Sry I didn't know that you can open 3mf files with cura

[image]

1800Coconut, 2022-02-15:
Okay that's strange:

When I save frvmrs uploaded gcode and try to print it, it works

When I save his 3mf file and try to print it, it doesn't work again, temperatures don't heat up

fvrmr, 2022-02-15:
Can you share your 3mf file(file - save project) that doesn't work? You need to share it as a .zip here on Github.

1800Coconut, 2022-02-15:
Alright, I took another file this time that also does not work

[CE3_Zähler.zip](https://github.com/Ultimaker/Cura/files/8068252/CE3_Zahler.zip)

1800Coconut, 2022-02-15:
Hmm I just tried something...

When I choose your file to print, the temperature starts to heat up

Then i stop the printing right after the start and choose one of my files that don't work

The temperature then starts to heat up on my file that usually doesn't work before

I really can print my files when I choosed your file before

1800Coconut, 2022-02-15:
After I press START on my file that does not work, there is no STOP option in the menue

It looks like the printer is missing something in the file

As if he does not recognize the start

fvrmr, 2022-02-25:
Hi @1800Coconut I just checked your 3mf and made a gcode file, but it seems like it is correct in your project file and the gcode that I created. 
So I think there is an issue with your printer firmware.

1800Coconut, 2022-02-25:
Well, I tried the Prusa slicer and the Creality slicer, both work fine with my printer now.
I guess I just can't User Cura anymore, but that's ok for me.

Thanks for your effort.

closed 2022-02-25 by 1800Coconut