Ultimaker/Cura #11465: Slicing error

MrReijii, 2022-02-11:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Flsun Super Racer

### Reproduction steps

When slicing the model, the upper face is not generated in narrow places (about 1-3 mm). This happens when choosing my printer Flsun Super Racer and also Anet A8. If you specify a printer of any Ultimaker, then there is no such problem with slicing. Sorry for my english
[image]
[image]
.

### Actual results

When printing such a model, there are holes on it, since there is simply no plane that needs to be printed there. In the attached files, the wall thickness is indicated in one layer specifically in order to better see the problem. Such a problem on any number of wall layers.
This problem was on previous versions of Sura, the update did not fix it. I removed the program completely. I deleted all profiles and printers. Installed the program on another PC and there is the same problem.
[image]
[image]
.

### Expected results

Same model but with
[image]
[image]
 selected printer Ultimaker S5

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[FSR_Korobka Litii B.zip](https://github.com/Ultimaker/Cura/files/8049205/FSR_Korobka.Litii.B.zip)

fvrmr, 2022-02-11:
Hi @MrReijii thank you for your report. 
You need to change the `Maximum Skin Angle for Expansion` to for example 90 °
[image]
This is probably set as default in your profile, you can change this setting and create your own profile, so you don't have this issue in the future. 
I will close this issue, but let me know if you have more questions.

closed 2022-02-11 by fvrmr

pixie-wrangler-irl, 2022-02-26:
Why did this value get changed in the default profile of only the Super Racer? What does it take to change the default back to 90 to save folks from this insanely niche issue.

Ghostkeeper, 2022-03-14:
See https://github.com/Ultimaker/Cura/issues/11553 for a fix to the FLSUN Super Racer profiles.