Ultimaker/Cura #11466: Thread Scheduling Issues on Intel 12th Gen CPUs

Leapo, 2022-02-11:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Deltesian

### Reproduction steps

1. Start Cura on a Windows 10 computer with an Intel 12th gen CPU that has both P cores and E cores enabled.
2. Start the Windows task manager.
2. Load any STL file into Cura.
3. Check the process priority and CPU utilization of CuraEngine.exe. Process priority will be "Below Normal", which results in the process running on E-cores only.

### Actual results

CuraEngine.exe sets a process priority of "Below Normal" for itself, which results in it being forced to run on E-cores only. This massively reduces slicing performance on Intel 12th gen processors.

### Expected results

CuraEngine.exe is expected set a process priority of "Normal", which allows it to use P cores if possible. This greatly improves slicing performance on 12th gen processors.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Manually setting the process priority for CuraEngine.exe to "Normal" temporarily works around the issue, but I would recommend implementing the same fix that Handbrake did, as they were having exactly the same problem on 12th gen CPUs: https://github.com/HandBrake/HandBrake/pull/3970

fvrmr, 2022-02-14:
Hi @Leapo thank you for your report. 
I will bring it up with the team. Keep you posted.

nallath, 2022-02-15:
Well, I hope that this might finally solve the issue that people have been reporting that Cura is slow for them (without us being able to reproduce it).

As there already is at least one project that got a bit faster due to this change (and the original decision for us to make it like it is was over 8 years ago) I don't see a lot of problem with changing this. 

For our reference: CURA-8957

closed 2022-02-24 by nallath