Ultimaker/Cura #11467: v4.12, v4.13, v4.13.1 Application crash (not responding)

Wobberjockey, 2022-02-11:
### Application Version

4.13.1

### Platform

Win 10 21H2 (19044.1526)

### Printer

Artillery Sidewinder x1

### Reproduction steps

Install Cura
set up printer using default X1 values.  (just hit next through the setup process)
load changes from account. 
wait 15 to 30  seconds.

### Actual results

right click in the build volume - crash
attempt to access menu - crash
change slicer settings - crash
load a model (any model file) - crash
try to do almost anything -> crash not responding

subsequent runs of cura will immediately go to a blank white screen, not responding after "loading build volumes"
issue will persist through OS restart (and even an OS update) until the application is completely uninstalled including all application date, user app data, and the application itself

clearing application is only a temporary fix 

### Expected results

the application should load. 
currently it seems like something in the sync to an ultimaker account is broken, and this process is corrupting some data preventing subsequent loads that said, issue is present even if account setup is skipped 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8052142/cura.log)

Issue is persisting through multiple cura versions, have been able to consistently reproduce on my system after every cura clean install
Chitubox has _**NEVER**_ been installed on the PC in question, and version 4.12.1 worked without issue until I updated to 4.13
All cura marketplace materials and plugins have been removed from my ultimaker account on the off chance that the issue was related  to something there.  issue is also present when skipping the account setup step. Issue has persisted through several graphics driver updates as well

fvrmr, 2022-02-14:
Hi @Wobberjockey thank you for your report.
Unfortunately your log file doesn't show what is happening and I can't reproduce your issue my self.
Also in the logs it shows that this has nothing to do with your Ultimaker account. 
Maybe this issue is related to yours? https://github.com/Ultimaker/Cura/issues/10905

Wobberjockey, 2022-02-14:
That does appear to be a similar problem, yes. 

I’ll test to see if there is a program running as admin in the background and see if that is the issue. (I think that was the thinking in the other thread, if I followed it properly) 

I had already tested agains fusion, Vivaldi (my browser), discord, and a few other apps but there might be something else that slipped under the radar. 

I do not believe I installed any other software around the time of the issue.

Wobberjockey, 2022-02-15:
hopefully double posting is not against guidelines

I did some testing of the items listed in #10905 

- Autodesk software - haven't run Fusion in ages, and not since prior to the issue
- curaCLI - lotts of spam here about USBPrinting.AutoDetectBaud.run.  issue seems to occur right after `[MainThread] UM.View.GL.ShaderProgram.load [60]: Loading [C:\Program Files\Ultimaker Cura 4.13.1\resources\shaders\xray.shader]...`
- High CPU use - it's hitting one core pretty hard. (11-12.5% of 8 cores) i do run process lasso, but the issue is unchanged after adding explicit exclusions for Cura and CuraCLI. Exiting process lasso entirely and disabling the process governor engine had no impact as well. 
- AV - i use bit defender and have not had issues before. added the entire c\program files\ultimaker cura\ directory as an exclusion. no change.
- Run as Admin - No change

Pleas let me know if you have any other ideas for testing

one possible lead

[image]

putting that through process explorer nets me this

[image]

fvrmr, 2022-02-22:
Hi @Wobberjockey do you maybe have a firewall enabled, are you using a VPN or do you have a network drive?

Wobberjockey, 2022-02-22:
Using the standard windows defender firewall. Cura is already as an exception, and allowed to communicate. 

There were no changes to the home modem/router that might block communication either. 

no VPN. 

No mapped network drives.

fvrmr, 2022-02-25:
Then we have to wait for other users running into this issue, who have reproduction steps which we can reproduce. Since we can't reproduce it ourselves or see what is causing this in the log file.

Wobberjockey, 2022-02-25:
Are you able to recommend any further steps I can take that might produce more data for you to work with?

Ghostkeeper, 2022-02-26:
The application crashes during the creation of the main window. Unfortunately, that main window creation is done by a GUI framework we're using, Qt. It's C++ code and doesn't log exceptions very well in the release build. So there is very little we can see from the crash.

One possibility to get more information is to run Cura through a debugger, such as [GDB](https://sourceware.org/gdb/) or Visual Studio Debugger. It'll give little information since Qt is a release build, but it'll at least indicate if the problem is in Qt, in Windows, in the GPU drivers or in some other component. Explaining how to run an application through GDB is sufficiently complex though (for someone not experienced with it) that I won't attempt to do that via Github comments.

So other than that, I hope that we can find some patterns in the people who experience this issue, if we get more reports about it.

fieldOfView, 2022-02-28:
Do you have multiple displays? If so, can you try running with a single display (ie: disconnect or disable the other display(s))?

Wobberjockey, 2022-03-02:
YES! on PC display only mode the program runs without any issue. fired right up and complained about not being able to contact the login server. 

Closing the program and re-opening it with dual displays caused the issue to reoccur instantly. More testing is required around this but it's clearly a factor - i will try playing with the graphics driver profile

Ghostkeeper, 2022-03-12:
Could be looking at https://bugreports.qt.io/browse/QTBUG-47059 or https://bugreports.qt.io/browse/QTBUG-65882 here.

tvixen, 2022-03-18:
I have the same issue as Wobberjockey.  Also having 2 screens. I have just run ver 4.8 and 4.10 these are running fine.

fieldOfView, 2022-03-18:
If Cura starts when you have all but a single screen disabled/disconnected, deselect the option to "Restore window position on start" in the General preferences. This makes sure that Cura always starts on the primary screen, which probably fixes this problem.

tvixen, 2022-03-18:
4.10 seem to work until I installed Sidebar GUI plugin from Marketplace. Then it stopped responding.

fieldOfView, 2022-03-18:
@tvixen I think that is merely coincidental. By default, Cura starts up in the position where you last closed it. On some systems, Cura will become "unresponsive" if it opens on the secondary screen. As long as Cura starts on the primary screen, there are probably no issues. That is what unselecting the option to "Restore window position on start" is for.

I think you "coincidentally" moved Cura 4.10 to the secondary screen, and then restarted Cura to finish installing the Sidebar GUI plugin.

Try disabling/disconnecting the second screen before opening Cura. If that works, be sure to uncheck "Restore window position on start".

tvixen, 2022-03-18:
I re-installed a 4.10 from scratch an start to turn on things one by one.
"Restore window position on start"  did not change the responce, but I ment something to set the "Setting Visibility" to ALL.  Then the laptop went crazy slow. And if I then unfold all the settings, (Infill, Material, Speed etc) then its almost not responding.

I will try this with the new version 13.1 also

tvixen, 2022-03-18:
Installed ver 4.10 and 4.13.1 on a Dell laptop. And no problems.
Re-installed version 4.10 and 4.13.1 on a HP laptop coulden't run 4.10 with Sidebar GUI (Without running fine)
And just did the same with 4.13.1 Totally dead (No responce) with sidebar GUI. But full responsive without Sidebar GUI plugin.

Sorry fieldOfView, but thats a fact.

tvixen, 2022-03-18:
Oki, did it one more time on my HP laptop, to prove that I was wrong.
Removed 4.13.1
Installed 4.13.1 clean install with out plugins
Everything was running 100% and no delays
installed my plugins from Marketplace except Sidebar GUI and did a restart of Cura.
Loaded a stl and sliced it.   No problems.
Then I installed Sidebar GUI, and restarted.
Cura Not responding  :( 
Deleted the profile in C:\Users\tim.milgart\AppData\Local\cura 4.13.1
and C:\Users\tim.milgart\AppData\Roaming\cura\4.13
Restarted Cure..... set it up again **without** Sidebar GUI.
Everything was working perfect again.

Wobberjockey, 2022-03-18:
> If Cura starts when you have all but a single screen disabled/disconnected, deselect the option to "Restore window position on start" in the General preferences. This makes sure that Cura always starts on the primary screen, which probably fixes this problem.

This seems to fix it for me

tvixen, 2022-03-19:
It didn't for me.  Even with both screens off, and only the laptop screen on, Cura was not responding after startup.
But when disabling the plugin it was responding again, and fast as usual.
I can't test/debug Cura while the Sidebar GUI plugin is enabled, as the program just show "Not Responding" after 1min.

But the HP laptop is the only pc I have this problem on (2 others is running fine). 
So I don't know if its interesting to debug the issue, fieldOfView?

closed 2024-12-09 by github-actions[bot]