Ultimaker/Cura #11468: Alternating Wall Line Counts

HMarxen, 2022-02-12:
### Is your feature request related to a problem?

The adhesion between an infill line meeting a wall line is often waeker than two paralel lines.

### Describe the solution you'd like

Having alternating wall line counts. E. g. every odd number of layers has a wall line count of 3, every even number of layers has a wall line count of 2. This would better squish in the ends of infill lines between two wall lines on top and bottom.

### Describe alternatives you've considered

none

### Affected users and/or printers

Everyone who needs strong parts.

### Additional information & file uploads

[image]

Asterchades, 2022-02-13:
Does the **Alternate Extra Wall** option (in the Walls section) no longer do this...?

HMarxen, 2022-02-13:
Oh damn. You are right. Sorry this is redundant then.

closed 2022-02-13 by HMarxen