Ultimaker/Cura #11469: nozzle size. filament temperature

statsevera, 2022-02-12:
### Application Version

4.13.1

### Platform

windows 10

### Printer

custom fff printers

### Reproduction steps

there is no change in nozzle size. the filament temperature does not change when selecting PLAY vs PETG

### Actual results

there is no change in nozzle size. the filament temperature does not change when selecting PLAY vs PETG

### Expected results

there is no change in nozzle size. the filament temperature does not change when selecting PLAY vs PETG

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

/

fvrmr, 2022-02-14:
Hi @statsevera thank you for your report. 
Could you explain a bit more what your issue is? Maybe with some screenshots.
Do you want to change the nozzle size?

statsevera, 2022-02-17:
I figured out the profile, the nozzle temperature is blocked in this profile. But the problem is that when choosing a material (PLA,Pet), the program does not report that it cannot change the temperature. And in the quick settings window, I can't see it at all. I need an error message and it seems to me that the icon of the blocked value is needed in red.
There is no possibility of changing the nozzle when adding "Custom FFF", it is clear that I can do this through setting up and creating my profiles. But why is there no filament selection in the line, the choice of the nozzle diameter as with other printers
[Hornet Normal.zip](https://github.com/Ultimaker/Cura/files/8087562/Hornet.Normal.zip)
?

statsevera, 2022-02-17:
If it is not clear, I will send a video)

fvrmr, 2022-02-25:
Hi @statsevera thank you for your profile. The reason why the temperature is not updated is because this is defined in your profile. 
`material_print_temperature = 210 ` so it will use 210 for all material types. If you change from PLA to PETG the temperature will not change, but you can click on the fx icon next to the temperature to update the temperature for PETG. 

Let me know if that works for you.

closed 2022-03-11 by no-response[bot]