Ultimaker/Cura #11470: MonitorPageProblem

BillMcClennen, 2022-02-12:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Ortur Obsidian

### Reproduction steps

Moved to monitor page
Tried to turn on "Preheat" for Extruder 1 and Build Plate after aborting print.

### Actual results

"Preheat" button not displayed.

### Expected results

The "Preheat" button should be displayed.
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

The log file and project file are not needed as this is a visual problem with the application.

fieldOfView, 2022-02-12:
Your screenshot shows your printer is currently printing. You cannot _pre_heat when you are already printing.

BillMcClennen, 2022-02-13:
I had aborted!

fieldOfView, 2022-02-13:
> I had aborted!

On your printer, or in Cura? When Cura is printing via USB, aborting the print on your printer will do nothing. Cura will not be aware of you aborting the print. In your screenshot, above the progress bar, you can still see that Cura is "Printing...".

BillMcClennen, 2022-02-13:
Aldo, thanks for the input.  I guess I was incorrect on this.  I’ll keep a look on it to see what caused me to be off.  If I find it, I’ll let you know.  You take care.

 

Bill

closed 2022-02-28 by no-response[bot]