Ultimaker/Cura #11471: Error after pause at height

SmallHauler, 2022-02-13:
### Application Version

4.13.1

### Platform

WINDOWS 10

### Printer

ender 3

### Reproduction steps

Added post processing script of pause at height.

### Actual results

Printer stopped as desired but when it heated back up and went to purge new material it first moved to the print and then deposited filament on the print.

### Expected results

When it heats back up it should first purge material and then move to the print.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Funky Bojo 2.zip](https://github.com/Ultimaker/Cura/files/8054781/Funky.Bojo.2.zip)

fvrmr, 2022-02-14:
Hi @SmallHauler thank you for your report.
What are you doing during the pause? Changing filament?
Because if so, it could be that the new filament is moved to far and after the pause the feedrate will move filament 45.0 mm
See this line in the gcode:
```
G1 F2700 E45
G1 F2700 ; restore extrusion feedrate
```

Ghostkeeper, 2022-02-26:
The Pause at Height script has functionality to purge some material after the pause. It is the Extrude Amount setting in that script. By default it is set to 0mm, but if set to e.g. 5mm, it'll add this right after the pause:

    G1 F199.998 E5; Extra extrude after the unpause

I think that is what you need. It is not a bug. You just didn't use the functionality that does what you want it to do.

SmallHauler, 2022-03-22:
I switched to an older version and it worked perfectly. It's not like I am the one creating the g code. Cura is doing it and it is doing it wrong

fvrmr, 2022-03-24:
Are you using the same project file?

SmallHauler, 2022-03-25:
No I re slice the original file

ydkjman, 2022-11-01:
> The Pause at Height script has functionality to purge some material after the pause. It is the Extrude Amount setting in that script. By default it is set to 0mm, but if set to e.g. 5mm, it'll add [...]

The problem is even when you don't check the box or "use" the functionality like you said, why does it move over the print and extrude a blog out onto the print and the starts printing. 
 You still have this issue even if you pick an amount to extrude before it goes to resume the  print.  
I picked an amount to extrude, it did extrude that amount but still went over the print and purged more out onto the print leaving that blob again.

SmallHauler, 2022-11-12:
I don't get it... I updated to the latest version and it ruined another print. There is no reason for it to go back to the print and then extrude filament. I am slicing it the same way I have in the past so I think if I am doing the same thing the only thing has changed is the program so there is so.ething wrong with Cura

GregValiant, 2024-11-16:
A change was made to the PauseAtHeight script to address this problem.  Is it still an issue or can this be closed?

closed 2024-11-20 by SmallHauler