Ultimaker/Cura #11472: Wall type changes depending on orientation (rotation angle z) of the part.

dtex91, 2022-02-13:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Voron 2.4

### Reproduction steps

For easyness i am using cura normal profile for this. Nothing modded.

I have loaded the .stl and wondered about hilarious walls. These walls are 0.4mm thick (approved it in fusion). 
Then i tried turning the whole stl by 45 degree on the z axis and the walls changed to nice ones. 
You can also see, that the walls line type changes...this seems to be a bug i guess. Maybe this behaviour creates problems somewhere else too.

[image]
[image]

### Actual results

Single walls at 0 deg rotation are ugly and not outer perimeter ones.
At 45 deg rotation the walls appear good and are treated like outer perimeter ones.

### Expected results

Indepentand of the rotation of the part, the walls should always be treated like outer perimeter ones.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Cura_Wall_problem.zip](https://github.com/Ultimaker/Cura/files/8054789/Cura_Wall_problem.zip)

GregValiant, 2022-02-13:
I had to drop down to .39mm Line Width to get it to slice.  I was able to turn off Print Thin walls.
[image]

fvrmr, 2022-02-14:
Hi @dtex91 thank you for your report. 
In Arachne this is not a problem anymore, since Arachne uses variable line width. See https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_beta_2
[image]

And since that will be the engine in our next Cura release, I will close this issue.

closed 2022-02-14 by fvrmr