Ultimaker/Cura #11473: Slicing Error in Cura 4.13.1 Leaving Gaps in Vertical Plane

davidlip-cura, 2022-02-13:
### Application Version

4.13.1

### Platform

Win 10

### Printer

Artillery X1

### Reproduction steps

I downloaded 4 .stl files from Maketory at https://www.thingiverse.com/thing:5199378 .

Layer height set at .2mm though changing to .4mm did not change the results.

### Actual results

It only sliced to layer 56 (out of 274) even though the model above that showed in grey.  Loaded the .stl into Cura 4.9 and it slices OK.

Scaled to 101%, 102%, 103% with no change.  104% got multiple segments separated by equal gaps.  Many White squares indicating Starts. Moving on up from 105% to 109% showed more and more printing still separated by large gaps.  

110% finally appears to slice the entire model.

Any scaling less than 100% does not scale the entire model.

Added something taller to the same project and the original nozzle still did not slice but the taller model did.

[image]

[image]

[image]

[image]

[5mm.zip](https://github.com/Ultimaker/Cura/files/8054941/5mm.zip)

``

### Expected results

I presume it should slice cleanly at 100%.  Odd that it slices in 4.9.0.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Nothing to add.  Printing now from slicing in 4.9.0.

GregValiant, 2022-02-13:
With the model loaded into 4.9 - select File / Save Project.  Load that project into 4.13.1 and see how it slices.  I didn't have any trouble in 4.13.1 so I think there is a setting in your 4.13.1 profile that is causing your problem.

GregValiant, 2022-02-13:
It might have been something like "Outer Wall Inset" but so long as it's fixed...

fvrmr, 2022-02-14:
Hi @davidlip-cura thank you for your report. 
Can your project file (from 4.13.1) in a .zip here on Github.